jkjung-avt/tensorrt_demos #424: Error making yolov4 trt int8

Hiperdyne19012, 2021-06-03:
[TensorRT] INTERNAL ERROR: Assertion failed: mRunner->mShader != nullptr
../builder/caskConvolutionBuilder.cpp:98
Aborting...
[TensorRT] VERBOSE: Builder timing cache: created 3162 entries, 15925 hit(s)
[TensorRT] ERROR: ../builder/caskConvolutionBuilder.cpp (98) - Assertion Error in setTactic: 0 (mRunner->mShader != nullptr)
ERROR: failed to build the TensorRT engine!

I get this. I am using CUDA 11.1, TRT7.2

jkjung-avt, 2021-06-03:
What GPU are you using?  Does it support INT8 inferencing?

Hiperdyne19012, 2021-06-03:
RTX 2080Ti. Yes there is INT8 support.

jkjung-avt, 2021-06-03:
> [TensorRT] INTERNAL ERROR: Assertion failed: mRunner->mShader != nullptr
> ../builder/caskConvolutionBuilder.cpp:98

Sorry, I have not seen this error before, neither do I find a match when I try to google it.

Previously, I have tested the INT8 code on my RTX-2080Ti (CUDA 10.2 & TensorRT 7.1.3.4) and it worked.  The only suggestion I have is to update your NVIDIA driver (or maybe try a different version of TensorRT & CUDA).

Hiperdyne19012, 2021-06-09:
The machine already has the latest NVIDIA GPU driver. I tried on a different machine with  RTX-2080Ti (CUDA 10.2 & TensorRT 7.2.3 (in the makefile I linked .so and include files extracted from tar but actual TensorRT installation was done using deb)) and it worked.

jkjung-avt, 2021-06-09:
> The machine already has the latest NVIDIA GPU driver.

I really don't have an idea.  Maybe you could post the issue onto NVIDIA Developer Forum...

closed 2021-06-17 by jkjung-avt