jkjung-avt/tensorrt_demos #427: Convert onnex to trt crashes the system.

maldivien, 2021-06-05:
This issue is consistent with some models.

I use AGX Xavier, might be a problem with power? I'm using max settings.

Can I convert to trt using a PC?

maldivien, 2021-06-05:
I'm using a yolov3 model, conversion to onnex was a success. Demos works as expected with very high frame rates so i dont think this a installation issue.

jkjung-avt, 2021-06-05:
Please use the "--verbose" (or "-v") command-line option and share the crash log.

maldivien, 2021-06-07:
Thanks. I have found the problem. It's due to insufficient/inconsistent power supply. AGX Xavier requires a very stable 65W power supply to run at MAX settings.

BTW I have question regarding NVDLA, what's the advantage of using it as it doesn't show any performance gains.

jkjung-avt, 2021-06-07:
> BTW I have question regarding NVDLA, what's the advantage of using it as it doesn't show any performance gains.

Please refer to https://github.com/jkjung-avt/tensorrt_demos/issues/364.

closed 2021-06-07 by maldivien