jkjung-avt/tensorrt_demos #431: load two models report errors

kaiven11, 2021-06-08:
I used a yolov3 trt engine  And a yolov5 trt engine, I tested  both of them  separately and everything is ok.

When put them togethor,Report the below error:

[image]

I did not change code in  the yolo_with_plugins.py ,
Only  flollow the https://github.com/jkjung-avt/tensorrt_demos/issues/203#issuecomment-680077788 ,change code in the yolov5_trt.py(this is my custom yolov5 file)

yolov5_trt.py:
`"""
An example that uses TensorRT's Python api to make inferences.
"""
import ctypes
import random
import threading
import os
import cv2
import numpy as np
import tensorrt as trt
import torch
import pycuda.driver as cuda
# import pycuda
import torchvision
from werkzeug.datastructures import ContentRange

INPUT_W = 608
INPUT_H = 608
CONF_THRESH = 0.1
IOU_THRESHOLD = 0.4

def plot_one_box(x, img, color=None, label=None, line_thickness=None):
    """
    description: Plots one bounding box on image img,
                 this function comes from YoLov5 project.
    param: 
        x:      a box likes [x1,y1,x2,y2]
        img:    a opencv image object
        color:  color to draw rectangle, such as (0,255,0)
        label:  str
        line_thickness: int
    return:
        no return

    """
    tl = (
        line_thickness or round(0.002 * (img.shape[0] + img.shape[1]) / 2) + 1
    )  # line/font thickness
    color = color or [random.randint(0, 255) for _ in range(3)]
    c1, c2 = (int(x[0]), int(x[1])), (int(x[2]), int(x[3]))
    cv2.rectangle(img, c1, c2, color, thickness=tl, lineType=cv2.LINE_AA)
    if label:
        tf = max(tl - 1, 1)  # font thickness
        t_size = cv2.getTextSize(label, 0, fontScale=tl / 3, thickness=tf)[0]
        c2 = c1[0] + t_size[0], c1[1] - t_size[1] - 3
        cv2.rectangle(img, c1, c2, color, -1, cv2.LINE_AA)  # filled
        cv2.putText(
            img,
            label,
            (c1[0], c1[1] - 2),
            0,
            tl / 3,
            [225, 255, 255],
            thickness=tf,
            lineType=cv2.LINE_AA,
        )

class YoLov5TRT(object):
    """
    description: A YOLOv5 class that warps TensorRT ops, preprocess and postprocess ops.
[... 235 lines cut ...]
if __name__ == "__main__":
    # load custom plugins
    PLUGIN_LIBRARY = "build/libmyplugins.so"
    ctypes.CDLL(PLUGIN_LIBRARY)
    engine_file_path = "build/yolov5x.engine"
    import pycuda.autoinit
    # load coco labels

    categories = ["smoking"]

    # a  YoLov5TRT instance
    yolov5_wrapper = YoLov5TRT(engine_file_path,1,0,pycuda.autoinit.context)

    # from https://github.com/ultralytics/yolov5/tree/master/inference/images
    input_image_paths = [ "12.jpg"]

    for input_image_path in input_image_paths:
        # create a new thread to do inference
        thread1 = myThread(yolov5_wrapper.detect, [input_image_path,0.5])
        thread1.start()
        thread1.join()

    # destroy the instance
    yolov5_wrapper.destroy()
`

jkjung-avt, 2021-06-08:
Please try the code snippet here: https://github.com/jkjung-avt/tensorrt_demos/issues/219#issuecomment-695429092

If it works, then the problem is likely due to either: 1. the TensorRT yolov5 engine is not built correctly, or, 2. bug in your code.

jkjung-avt, 2021-06-08:
I have no idea about "./build/libmyplugins.so".  I would have checked if there are name conflicts between the 2 so files...

kaiven11, 2021-06-08:
sorry, I follow the  https://github.com/wang-xinyu/tensorrtx/tree/yolov5-v3.0/yolov5  to gender yolov5.trt file and  the  .so file .

I just copy the file to this project.

jkjung-avt, 2021-06-08:
As stated earlier, you'll have to resolve naming conflicts between the 2 .so files by yourself.

kaiven11, 2021-06-08:
I edit the two  yolo_layer.cu file,  one for yolov3,another for yolov5 , 
 1.change the namespace  yolo3  and yolo5  , 
 2. one change  YoloLayerPlugin to YoloLayerPlugin3 , another YoloLayerPlugin5
 3. make success
 4. test separately (success)
 5. test together (fail)
 
error message:
[image]

Maybe can't init the nvinfer1 two times?
Pls give me some advice, Thank you.

jkjung-avt, 2021-06-08:
The 2 plugins cannot register the same name and version.  After you modify plugin name of one of them, you need to re-build the TensorRT engine with the updated name.

https://github.com/jkjung-avt/tensorrt_demos/blob/bac5085d75e59369cc587cf9ec19f5120836cb3e/plugins/yolo_layer.cu#L337-L345

closed 2021-06-09 by jkjung-avt