jkjung-avt/tensorrt_demos #432: Getting issue while doing make in plugins

arslanfirat, 2021-06-10:
Hi ,
I am getting the follwoing error while doing make in plugins folder. Can please help to resolve this error?

`computes: 72
NVCCFLAGS: -gencode arch=compute_72,code=[sm_72,compute_72]
nvcc -ccbin g++ -I"/usr/local/cuda/include" -I"/usr/local/TensorRT-7.1.3.4/include" -I"/usr/local/include" -I"plugin" -gencode arch=compute_72,code=[sm_72,compute_72] -Xcompiler -fPIC -c -o yolo_layer.o yolo_layer.cu
g++ -shared -o libyolo_layer.so yolo_layer.o -L"/usr/local/cuda/lib64" -L"/usr/local/TensorRT-7.1.3.4/lib" -L"/usr/local/lib" -Wl,--start-group -lnvinfer -lnvparsers -lnvinfer_plugin -lcudnn -lcublas -lcudart_static -lnvToolsExt -lcudart -lrt -ldl -lpthread -Wl,--end-group
/usr/bin/ld: cannot find -lcudart_static
collect2: error: ld returned 1 exit status
Makefile:34: recipe for target 'libyolo_layer.so' failed
make: *** [libyolo_layer.so] Error 1
`

jkjung-avt, 2021-06-10:
> /usr/bin/ld: cannot find -lcudart_static

The "-lcudart_static" could be removed from the Makefile.

I just commit the update to the repo.  You could `git pull` and try again.

arslanfirat, 2021-06-10:
It works!! thanks you very much

closed 2021-06-10 by arslanfirat