jkjung-avt/tensorrt_demos #433: make does not work

Myron1996, 2021-06-14:
Hi @jkjung-avt . I am trying to run `make` command in plugins directory but I am receiving following error .  Can you please advice me on this ? Thank you 
```
Makefile:20: "compute=75" is for GeForce RTX-2080 Ti.  Please make sure CUDA compute is set correctly for your system in the Makefile.
g++ -shared -o libyolo_layer.so yolo_layer.o -L"/usr/local/cuda/lib64" -L"/usr/local/TensorRT-7.1.3.4/lib" -L"/usr/local/lib" -Wl,--start-group -lnvinfer -lnvparsers -lnvinfer_plugin -lcudnn -lcublas -lcudart_static -lnvToolsExt -lcudart -lrt -ldl -lpthread -Wl,--end-group
/usr/bin/ld: cannot find -lnvparsers
/usr/bin/ld: cannot find -lnvinfer_plugin
collect2: error: ld returned 1 exit status
Makefile:45: recipe for target 'libyolo_layer.so' failed
make: *** [libyolo_layer.so] Error 1

```

jkjung-avt, 2021-06-14:
> /usr/bin/ld: cannot find -lnvparsers
> /usr/bin/ld: cannot find -lnvinfer_plugin

This means "ld" was not able to find your TensorRT libraries ("libnvparsers.so" and "libnvinfer_plugin").  You should check your TensorRT installation.

Myron1996, 2021-06-14:
Sure , I thought the same. Thank you Bro. I will look in to it.

Myron1996, 2021-06-14:
It was tensorrt installation fault . It's been resolved

closed 2021-06-14 by Myron1996

rafaelbate, 2021-06-18:
> It was tensorrt installation fault . It's been resolved

Hello @Myron1996, having the same issue when trying to install in Google Colab .. Could you tell me what was your fix? Where is tensorrt located? Thanks :)

Myron1996, 2021-06-18:
Hi @rafaelbate . I actually used a docker image which has all the dependency's installed .I was working on local machine. 

You can check following repo. It will show you how to install the docker image. I hope it helps 

https://github.com/GeekAlexis/FastMOT/blob/master/README.md

rafaelbate, 2021-06-22:
Thanks @Myron1996, I got it running using WSL2 and docker as you mentioned :)