jkjung-avt/tensorrt_demos #434: Detections in images, but no detections on video.

DJT777, 2021-06-19:
Hi,

I've trained a new model for tiny-yolov4. It runs trt_yolo after converting it to onnx and trt. However I am getting almost 0 detections above a  50%-60% threshold, and most of the detections that occure below that threshold are not accurate at all.

I am wondering if you may help me figure out why the model performs well on images, but is not doing so well on video.

https://media.discordapp.net/attachments/790297894316933150/855822261611921448/Screenshot_from_2021-06-19_09-35-06.png

https://media.discordapp.net/attachments/790297894316933150/855822288985391144/birds1.png

jkjung-avt, 2021-06-20:
I'm not able to identify the problem by looking at the pictures.

Could you capture a few image frames from the video and try inferencing those instead?

DJT777, 2021-06-21:
@jkjung-avt 

I think the issue was in what data the model was trained on. It was trained on single images of birds from the NA-Birds dataset making the model only recognize similar positions of the birds therein.

Here are videos for sampling

One video demonstrates great results, and the other shows that it is making many detections that are not accurate. 

https://youtu.be/ymqvseLPFrI

https://youtu.be/TkUgokBB1o4

I think I found the solution: To show the camera and model angles similar to what it was trained on.

Here are sample images of the dataset:
https://imgur.com/a/lzOKWYc

Please let me know if you think it may be something else.

jkjung-avt, 2021-06-21:
> It was trained on single images of birds from the NA-Birds dataset

I think you are going towards the right direction.

I'd also suggest you to consider the following:

* Make sure the types of birds you're trying to detect are covered in the training set.
* Make sure the camera angle/distance of the videos is similar to what appears in the training set.
* Try some data augmentation (adding noise or perturbing the colors) to compensate difference in camera characteristics between training and test sets.
* Birds in the videos might be in motion (or even blurred), which are not present in the still image training set.  Again, you might try some data augmentation to force the model to learn that.

DJT777, 2021-06-21:
@jkjung-avt 

Thank you for the suggestions.

closed 2021-06-29 by jkjung-avt