jkjung-avt/tensorrt_demos #435: how to filter the cocodataset classes that I want to be detected

jmudy, 2021-06-20:
I would like to know how to filter the cocodataset classes that I want to be detected. In my case, I am trying to detect only cars in `trt_yolo.py` script and I don't know which variable I have to filter in the detection.

Thanks in advance.

jkjung-avt, 2021-06-21:
https://github.com/jkjung-avt/tensorrt_demos/blob/549d50b167f57f121e2fc8eed214a30d41f19de3/trt_yolo.py#L66

The outputs of TrtYOLO.detect() are:

* boxes: bounding box coordinates
* confs: confidence scores
* clss: class id's of the detections

So you could add just code to do filtering based on `clss`.

You could also try the following code snippet to better understand the above.

```
import cv2
import pycuda.autoinit
from utils.yolo_with_plugins import TrtYOLO
img = cv2.imread('dog.jpg')
trt_yolo= TrtYOLO('yolov4-416', (416, 416), 80)
print(trt_yolo.detect(img))
```

Output:  `(boxes, confs, clss)`

```
(array([[132, 230, 309, 541],
       [124, 135, 574, 410],
       [465,  78, 692, 170]]), array([0.9861156 , 0.98989975, 0.9101637 ], dtype=float32), array([16.,  1.,  7.], dtype=float32))
```

Class ID 16 is "dog", 1 is "bicycle" and 7 is "truck".

[image]

closed 2021-06-21 by jmudy

philongvn99, 2022-03-12:
Excuse me for re-open this issue but if I want to have a result like [Demo#5 of Avenger Video](https://github.com/jkjung-avt/tensorrt_demos#demo-5-yolov4), is there now anyway like configuring then rebuild **.trt** file, I think **filtering the detection** will spend more time. Thank you for help!

jkjung-avt, 2022-03-14:
You could train a custom model with only object classes you want to detect.  For example, the Avengers video was done using a custom model ([jkjung-avt/yolov4_crowdhuman](https://github.com/jkjung-avt/yolov4_crowdhuman)) which only detects 2 object classes: "head" and "person".