jkjung-avt/tensorrt_demos #436: nvcc fatal   : Unsupported gpu architecture 'compute_86'

Maryam69, 2021-06-21:
Hi Jkjung
Thanks for the updating this NVIDIA project in python3. That is very useful. 
Actually, I could create yolov3-416.onnx file. After that, when I want to convert .onnx file to TensorRt, I am faced to this error **"ERROR: failed to load ../plugins/libyolo_layer.so.  Did you forget to do a "make" in the "../plugins/" subdirectory?"**. So, I explore to enter Folder "plugins/" and then run "make" command. But , unfortunately, this error was shown to me **"nvcc fatal   : Unsupported gpu architecture 'compute_86'".** 
My gpu is GeForce RTX 3090.
How to solve this problem? Do I run this code on this type of gpu, It is necessary to say that I run this git on the tensorrt:19.10-py3 NVC container.
Best,
Mary

jkjung-avt, 2021-06-21:
Referring to https://developer.nvidia.com/cuda-gpus, CUDA compute capability of GeForce RTX 3090 is "8.6".  So "compute_86" matches the GPU you are using.

I think the problem is that you are using an old version of CUDA Toolkit which does not support "compute_86" yet.  You should update your CUDA Toolkit to the latest version and try again.

Refer to https://docs.nvidia.com/deploy/cuda-compatibility/index.html for more details.

closed 2021-06-29 by jkjung-avt

Maryam69, 2021-07-01:
Thanks a lot Jung, You are right. This problem was related to the Cuda
version.
Best,
Mary🌹👍

shivamordanny, 2021-07-26:
Hi @Maryam69 , I am facing the same issue with RTX-3080, Can you please let me know which CUDA version, TensorRT version are compatible with compute_86 machines?
I am using the following container:
https://docs.nvidia.com/deeplearning/tensorrt/container-release-notes/rel_20-09.html#rel_20-09

Can you point me the one which works!

Maryam69, 2021-07-27:
Hi @shivamordanny 
Acording to this link  [https://developer.nvidia.com/cuda-gpus](url), all type of GeForce products are compatible with "compute_86". please check this link and select "GeForce and TITAN Products". 
If you want to find the suitable CUDA version, please check Table3 in this link [https://docs.nvidia.com/deploy/cuda-compatibility/index.html](url). Moreover, this command "dpkg -l | grep nvinfer" helps you to check cuda and TensorRT version.
In addition, you need to install onnx (1.4.1) correctly.
use these commands for installing:
"pip install --upgrade onnx==1.4.1"
"apt-get install protobuf-compiler libprotoc-dev"
I hope these suggestions help you to solve your problem.
Best,
Mary

shivamordanny, 2021-07-27:
Hi @Maryam69 , thanks for the refrences.
I used the recent tensorrt image and it worked out well.

Maryam69, 2021-07-27:
Glad to hear it! 👍