jkjung-avt/tensorrt_demos #437: issue with building the modnet demo 'getPluginCreator could not find plugin InstanceNormalization_TRT version 1'

Tetsujinfr, 2021-06-21:
hi
I am on xavier NX jetpack 4.4DP (so tensorrt 7.1)

I have build successfully (I think) all the steps for Modenet until this one:
`LD_LIBRARY_PATH=$(pwd)/onnx-tensorrt/build onnx-tensorrt/build/onnx2trt modnet.onnx -o modnet.engine -d 16`

then I got an error message:
```
ERROR] INVALID_ARGUMENT: getPluginCreator could not find plugin InstanceNormalization_TRT version 1
While parsing node number 219 [InstanceNormalization -> "721"]:
--- Begin node ---
input: "718"
input: "719"
input: "720"
output: "721"
name: "InstanceNormalization_219"
op_type: "InstanceNormalization"
attribute {
  name: "epsilon"
  f: 1e-05
  type: FLOAT
}

--- End node ---
ERROR: /media/tetsfr/SSD/tensorrt_demos/modnet/onnx-tensorrt/builtin_op_importers.cpp:1636 In function importInstanceNormalization:
[8] Assertion failed: plugin != nullptr && "InstanceNormalization plugin was not found in the plugin registry!"
```

Do you what that could be related to?

jkjung-avt, 2021-06-22:
> [ERROR] INVALID_ARGUMENT: getPluginCreator could not find plugin InstanceNormalization_TRT version 1

It looks like the "InstanceNormalization_TRT" plugin is missing (from /usr/lib/aarch64-linux-gnu/libnvinfer_plugin.so).  This is most likely because InstanceNormalization_TRT is not included in older builds of TensorRT.

I myself have tested the MODNet code against both JetPack-4.5 and JetPack-4.5.1.  Would you be able to use one of those instead?

Tetsujinfr, 2021-06-22:
Ok thanks, I ll try your great code once I can install Jetpack 4.5, but for now I am stuck with 4.4DP, too many dependencies on this version.
Closing the issue since root cause and solution to fix are clear. Thanks again for your amazing repository!

closed 2021-06-22 by Tetsujinfr