jkjung-avt/tensorrt_demos #438: trt_yolo len(engine) = 7 , ERROR：assert 3 <= len(engine) <= 4 how to solve it ?

Hansansui-cv, 2021-06-22:


jkjung-avt, 2021-06-22:
How many yolo layers are there in your custom yolo model?

Hansansui-cv, 2021-06-23:
> How many yolo layers are there in your custom yolo model?

well, in fact, i do not understand what is the engine mean. And, i am using the model nanodet with your code. I think it would not have any question, but i am wrong. So, the engine means layer? Besides, i have print the len(engine), got the number 7. I want to know how to revise the code? I need your help. Thank you.

jkjung-avt, 2021-06-23:
> well, in fact, i do not understand what is the engine mean.

Please read TensorRT documentation.

> Besides, i have print the len(engine), got the number 7.

This means the TensorRT engine has 7 input/output "bindings" in total.

Hansansui-cv, 2021-06-23:
> > well, in fact, i do not understand what is the engine mean.
> 
> Please read TensorRT documentation.
> 
> > Besides, i have print the len(engine), got the number 7.
> 
> This means the TensorRT engine has 7 input/output "bindings" in total.

OK, thank you

closed 2021-06-29 by jkjung-avt