jkjung-avt/tensorrt_demos #440: mtcnn detection poor fr heads pointing down

Tetsujinfr, 2021-06-23:
hi, is it a know fact that mtcnn is not robust to rotations? the landmarks are consistently detected for heads on top of bodies but if the heads are pointing down, then it is all over the place (either does not detect anything or put yes where mouth should be and vice versa).

Good detection exemple below:
[image]

Poor examples below:
[image]

[image]

[image]

jkjung-avt, 2021-06-23:
> hi, is it a know fact that mtcnn is not robust to rotations?

I did not train this model, neither did I verify such cases.  I think this is likely due to lack of upside-down faces in the training set.

Tetsujinfr, 2021-06-23:
Ok thanks. Was curious to see if there was any experience with such scenarios which I did not expect tbh. Likely training set related indeed.
Closing

closed 2021-06-23 by Tetsujinfr