jkjung-avt/tensorrt_demos #442: /usr/bin/ld: cannot find -lboost_python*-py3?

Sam-Fisher-20, 2021-06-28:
Hi - my python3.8 is in these places

python3: /usr/bin/python3 /usr/bin/python3.8-config /usr/bin/python3.8 /usr/lib/python3 /usr/lib/python3. 9 /usr/lib/python3.8 /etc/python3 /etc/python3.8 /usr/local/lib/python3.8 /usr/include/python3.8 /usr/sha re/python3 /usr/share/man/man1/python3.1.gz

I reinstalled python3.8 (thinking it would resolve -lboost error) 
Then deleted pycuda tree and ran install_pycuda script again. 
I got the error thrown at me again. 

/usr/bin/ld: cannot find -lboost_python*-py3?
collect2: error: ld returned 1 exit status
error: command 'x86_64-linux-gnu-g++' failed with exit status 1
make: *** [Makefile:6: all] Error 1

Any suggestion would be great. Thanks foryour help

Sam-Fisher-20, 2021-06-28:
The real issue is I am not able to locate -lboost_python*-py3?

jkjung-avt, 2021-06-28:
Could you do the following and see which .so files show up on your system?

```
$ ls /usr/lib/x86_64-linux-gnu/libboost_python*
```

Sam-Fisher-20, 2021-06-29:
> This Message originated outside your organisation.

Hi Jung

There were libboost_python38.so files and I tried to create soft link with libboost_python38-py3.so- that did not work
I used ln -s XXXXXX.so XXXXXX.so

I sorted it by removing your tree and just sudo apt install pycuda directly.

Thank you

Emails and their contents may be monitored for the purposes of quality control, fact checking and training.
This email, together with any attachments, is confidential and may be privileged.  If you have received it in error, please notify the sender and delete it.
To view the TRL Privacy Notice please visit https://trl.co.uk/permanent-landing-pages/privacy-notice.
TRL Limited, registered in England, No. 3142272, registered office: Crowthorne House, Nine Mile Ride, Wokingham, RG40 3GA, UK. VAT Registration 664 625 321.

jkjung-avt, 2021-07-01:
> I sorted it by removing your tree and just sudo apt install pycuda directly.

That should work.

closed 2021-07-01 by jkjung-avt