jkjung-avt/tensorrt_demos #444: Support for batch_size>1 for TensorRT (YOLO demo)

kgksl, 2021-06-30:
@jkjung-avt thanks alot for the great work. I am planning to try to convert my YOLOv4 models to TensorRT with your scripts. Havent started working on it yet. Was wondering if you plan to support specifying the batch size when converting darknet models to ONNX /converting ONNX model to TensorRT.
If not I will try to modify your scripts to achieve that (and will post if I have questions). Thanks alot again.

jkjung-avt, 2021-06-30:
@kgksl  The current code could work for batched TensorRT yolo inference with just some minor tweaks.  You could refer to https://github.com/jkjung-avt/tensorrt_demos/issues/373, for example.

I'm hesitant to officially add batched inference support into this repo, because it would complicate the code quite a bit.  Also, it's challenging to handle multiple video inputs efficiently (considering synchronization of image frames and dealing with dropped frames and other errors).

kgksl, 2021-06-30:
@jkjung-avt Thanks alot for the quick response and thanks again for the great work!