jkjung-avt/tensorrt_demos #445: Demo #5: YOLOv4 - "yolo_layer" plugin - can we skip building it?

kgksl, 2021-06-30:
Hi. I read in the readme that "yolo_layer" plugin is there to help speed up inference time of the yolov3/yolov4 models. If thats possible, could you please explain a bit more about this? I assume its used when we do ONNX to TensorRT conversion?

Can we skip build the "yolo_layer" plugin, and using 1onnx_to_tensorrt.py` and  use `.trtexec` to do ONNX to TensorRT conversion instead? ( I am planning to use this with low end NVIDIA GPU like GTX 1060 and not Jetson Nano board). I thought this might let me avoid any potential issues with TensorRT8 (https://github.com/jkjung-avt/tensorrt_demos/issues/425)? 

Thanks alot.

kgksl, 2021-06-30:
sorry for my previous question. Just realised `"yolo_layer" plugin` is used by `trt_yolo.py`. Anyway if possible could you please explain its purpose? thank you

jkjung-avt, 2021-06-30:
The "yolo_layer" plugin is needed.  "trt_yolo.py" would not work if you don't build it.

I have yet to update my "yolo_layer" plugin code for TensorRT 8 API.  For the time being, you could try this pull request submitted by @y-okumura-isp:
https://github.com/jkjung-avt/tensorrt_demos/pull/426

kgksl, 2021-06-30:
@jkjung-avt thank you for your quick response.

kgksl, 2021-07-02:
@jkjung-avt @y-okumura-isp 
I tired running YOLOv4-416 with tensorRT8 (tensorRT 8.0.0.3 and CUDA 11.3) using the pull request #426 and it worked. Thanks alot for the great work!

For YOLOv4-416, mAP @IoU=0.5:0.95 was 0.425 and mAP @IoU=0.5 was 0.652 . Reported values in repo were 0.459 and 0.700. Not sure if thats expected due to the differences that can happen when  TensorRT model  generation/optimisation according to different machines.

jkjung-avt, 2021-07-02:
You could refer to [README_mAP.md](https://github.com/jkjung-avt/tensorrt_demos/blob/master/README_mAP.md) for how I evaluated mAP of the TensorRT (COCO 80-classes) models.  I don't know why you get different numbers from me...

kgksl, 2021-07-03:
@jkjung-avt I found the issue that was causing difference between the reported mAP and the mAP I got.

In [`eval_yolo.py`](https://github.com/jkjung-avt/tensorrt_demos/blob/master/eval_yolo.py#L104) the model is defined as follows: 
`trt_yolo = TrtYOLO(args.model, (h, w), args.category_num, args.letter_box)`
but [`__init__` of TrtYOLO](https://github.com/jkjung-avt/tensorrt_demos/blob/master/utils/yolo_with_plugins.py#L274) doesnt expect `(w,h)` arguments.
`def __init__(self, model, category_num=80, letter_box=False, cuda_ctx=None)`
but since we provide it when we initialize model, letter_box is set to 80, and since `if 80` is True, the preprocessing and postprocessing of yolov4 model happens with `letter_box = True` although we expect it to be false. 

I found this because I got an error when trying to run yolov4-csp with `-l` flag. 

After I fixed it (removed `(w,h)` arguments in eval_yolo.py,  I get reported results for yolov4-416.
Thanks.

jkjung-avt, 2021-07-03:
Ahhhh, thanks for pointing out the bug.  I forgot to update "eval_yolo.py" when I committed https://github.com/jkjung-avt/tensorrt_demos/commit/52a699dac94041b28074179d093be6e5669c8741.

It is now fixed in the repo.

closed 2021-07-07 by jkjung-avt

kgksl, 2021-08-10:
@jkjung-avt just for my understanding, is there a reason why the `yolo layer` post processing part cannot be included as a onnx node? (Onnx python API doesnt support?) or is it just the way you decide to do? Thanks alot

jkjung-avt, 2021-08-10:
"yolo_layer" is implemented as a TensorRT plugin since it's not an [operation supported by TensorRT](https://docs.nvidia.com/deeplearning/tensorrt/support-matrix/index.html#layers-matrix).

It is not a [supported ONNX op](https://github.com/onnx/onnx/blob/master/docs/Operators.md) either.  So I don't include it in the ONNX model.  Instead, I add the plugins manually before building the TensorRT engine.

kgksl, 2021-08-10:
got it. thank you very much for the quick reply.