jkjung-avt/tensorrt_demos #446: I get a PyCUDA error when starting program at boot.

MarritS, 2021-07-03:
When I run the program manually everything is fine, however when I attempt to start it at boot using crontab get the following error:

`PyCUDA ERROR: The context stack was not empty upon module cleanup.

A context was still active when the contest stack was being cleaned up. At this point in our execution, CUDA may already have been deinitialized, so there is no way we can finish cleanly. The program will be aborted now. 
Use Context.pop() to avoid this problem`

jkjung-avt, 2021-07-04:
This is usually an error resulted from some other (prior) errors in the code.  Could you provide a full error log instead?

jkjung-avt, 2021-07-10:
Closing due to no response.

closed 2021-07-10 by jkjung-avt

WackyHaker, 2022-10-12:
Hi, I am running into the same problem here. Did @MarritS manage to resolve this issue? I also have a log here, if @jkjung-avt can still assist me. 
[image]