jkjung-avt/tensorrt_demos #451: No Detections

Myron1996, 2021-07-17:
HI @jkjung-avt  . I hope you are doing well. I have a quick question 

I am running a custom trained  YOLOV4-Tiny Model with 12 classes. 

I managed to convert the model in to `**.trt** ` but When I am running an inference it is not showing  any detection's on video. Do you have any input on that ?

### Environment 
TensorRT Version: 7.1.3
Hardware  - NVIDIA Xavier 
Python Version (if applicable): 3.6.9

jkjung-avt, 2021-07-18:
Have you verified your model with darknet?

Otherwise, check if this helps: https://github.com/jkjung-avt/tensorrt_demos/issues/434

Myron1996, 2021-07-21:
HI @jkjung-avt , Thank you so much for above advice. apparently, reinstalling the docker image got everything working. :+1:

closed 2021-07-21 by Myron1996