jkjung-avt/tensorrt_demos #455: Yolo to ONNX - TypeError: swish activation not supported

levipereira, 2021-07-23:
Hi,
There is plan to support swish activation?

```
$ python3 yolo_to_onnx.py -m yolov4-csp-swish-640 -c 5

Parsing DarkNet cfg file...
Building ONNX graph...
Traceback (most recent call last):
  File "yolo_to_onnx.py", line 991, in <module>
    main()
  File "yolo_to_onnx.py", line 976, in main
    yolo_model_def = builder.build_onnx_graph(
  File "yolo_to_onnx.py", line 516, in build_onnx_graph
    major_node_specs = self._make_onnx_node(layer_name, layer_dict)
  File "yolo_to_onnx.py", line 589, in _make_onnx_node
    node_creators[layer_type](layer_name, layer_dict)
  File "yolo_to_onnx.py", line 758, in _make_conv_node
    raise TypeError('%s activation not supported' % layer_dict['activation'])
TypeError: swish activation not supported
```

jkjung-avt, 2021-07-24:
I might look into it when I have time.  The swish activation is not supported for now.

jkjung-avt, 2021-08-08:
@levipereira  I just add code to support swish activation with the [9dd56b](https://github.com/jkjung-avt/tensorrt_demos/commit/9dd56b3b8d841dcfc2e5d1868f4bd785a50cbe98) commit.  Could you suggest a model for me to test the code?  Or maybe you could help to test it and let me know if it works.

levipereira, 2021-08-08:
@jkjung-avt I sent in your email google drive link to download a model to test. Let me know if links worked.

jkjung-avt, 2021-08-09:
@levipereira  I'm able to build a TensorRT engine from the model you provided.  And the TensorRT engine could be used to detect vehicles in images.  I'm not able to check mAP difference between the original DarkNet model and the TensorRT engine, though.

I think you could give it a try.

levipereira, 2021-08-11:
Many tanks.. will check mAP.

closed 2021-08-24 by jkjung-avt