jkjung-avt/tensorrt_demos #459: Makefile:37: recipe for target 'yolo_layer.o' failed

Enso-Zheng, 2021-08-09:
Hi ,
I am getting the follwoing error when doing make in plugins folder
and i try the sloution of https://github.com/jkjung-avt/tensorrt_demos/issues/400
but it's not work.

:~/tensorrt_demos/plugins$ make all -j
computes: 72
NVCCFLAGS: -gencode arch=compute_72,code=[sm_72,compute_72]
nvcc -ccbin g++ -I"/usr/local/cuda/include" -I"/usr/local/TensorRT-7.1.3.4/include" -I"/usr/local/include" -I"plugin" -gencode arch=compute_72,code=[sm_72,compute_72] -Xcompiler -fPIC -c -o yolo_layer.o yolo_layer.cu
yolo_layer.h(44): error: not a class or struct name

yolo_layer.h(50): error: member function declared with "override" does not override a base class member

yolo_layer.h(52): error: identifier "IPluginV2IOExt" is undefined

yolo_layer.h(52): error: member function declared with "override" does not override a base class member

yolo_layer.h(54): error: member function declared with "override" does not override a base class member

yolo_layer.h(56): error: member function declared with "override" does not override a base class member

yolo_layer.h(58): error: member function declared with "override" does not override a base class member

yolo_layer.h(60): error: member function declared with "override" does not override a base class member

yolo_layer.h(62): error: member function declared with "override" does not override a base class member

yolo_layer.h(64): error: member function declared with "override" does not override a base class member

yolo_layer.h(66): error: member function declared with "override" does not override a base class member

yolo_layer.h(68): error: member function declared with "override" does not override a base class member

yolo_layer.h(70): error: identifier "PluginTensorDesc" is undefined

yolo_layer.h(70): error: member function declared with "override" does not override a base class member

yolo_layer.h(72): error: member function declared with "override" does not override a base class member

yolo_layer.h(74): error: member function declared with "override" does not override a base class member

yolo_layer.h(76): error: member function declared with "override" does not override a base class member

yolo_layer.h(78): error: member function declared with "override" does not override a base class member

yolo_layer.h(80): error: member function declared with "override" does not override a base class member

yolo_layer.h(82): error: member function declared with "override" does not override a base class member

yolo_layer.h(84): error: member function declared with "override" does not override a base class member

yolo_layer.h(86): error: member function declared with "override" does not override a base class member

yolo_layer.h(89): error: identifier "PluginTensorDesc" is undefined

yolo_layer.h(89): error: identifier "PluginTensorDesc" is undefined

yolo_layer.h(89): error: member function declared with "override" does not override a base class member

yolo_layer.h(91): error: member function declared with "override" does not override a base class member

yolo_layer.h(96): error: member function declared with "override" does not override a base class member

yolo_layer.h(70): error: name followed by "::" must be a class or namespace name

yolo_layer.h(127): error: identifier "IPluginV2IOExt" is undefined

yolo_layer.h(127): error: return type is not identical to nor covariant with return type "nvinfer1::IPluginV2 *" of overridden virtual function "nvinfer1::IPluginCreator::createPlugin"

yolo_layer.h(127): error: member function declared with "override" does not override a base class member

yolo_layer.h(129): error: identifier "IPluginV2IOExt" is undefined

yolo_layer.h(129): error: return type is not identical to nor covariant with return type "nvinfer1::IPluginV2 *" of overridden virtual function "nvinfer1::IPluginCreator::deserializePlugin"

yolo_layer.h(129): error: member function declared with "override" does not override a base class member

/usr/include/aarch64-linux-gnu/NvInfer.h(5165): error: object of abstract class type "nvinfer1::YoloPluginCreator" is not allowed:
            pure virtual function "nvinfer1::IPluginCreator::createPlugin" has no overrider
            pure virtual function "nvinfer1::IPluginCreator::deserializePlugin" has no overrider
          detected during instantiation of class "nvinfer1::PluginRegistrar<T> [with T=nvinfer1::YoloPluginCreator]" 
yolo_layer.h(147): here

yolo_layer.cu(76): error: identifier "IPluginV2IOExt" is undefined

yolo_layer.cu(293): error: identifier "IPluginV2IOExt" is undefined

yolo_layer.cu(363): error: identifier "IPluginV2IOExt" is undefined

38 errors detected in the compilation of "/tmp/tmpxft_00002484_00000000-6_yolo_layer.cpp1.ii".
Makefile:37: recipe for target 'yolo_layer.o' failed
make: *** [yolo_layer.o] Error 1

Could you give me a hand, please？

jkjung-avt, 2021-08-09:
Please check out [README_x86.md](https://github.com/jkjung-avt/tensorrt_demos/blob/master/README_x86.md).  You probably need to modify `TENSORRT_INCS` and `TENSORRT_LIBS` in your "plugins/Makefile".

akyola, 2021-08-10:
Hi, I am getting the same error but i was trying on a Jetson Nano developer Kit not on ax86_64 PC. Do you have any idea how to solve this. I tried to find the TensortRT paths so that  i could try to change the TENSORRT_INCS and TENSORRT_LIBS in the Makefile but i couldn't although i am sure TensorRT 7.1.3.0 is installed. Below is the output of make.

computes: 53
NVCCFLAGS: -gencode arch=compute_53,code=[sm_53,compute_53]
nvcc -ccbin g++ -I"/usr/local/cuda/include" -I"/usr/local/TensorRT-7.1.3.0/include" -I"/usr/local/include" -I"plugin" -gencode arch=compute_53,code=[sm_53,compute_53] -Xcompiler -fPIC -c -o yolo_layer.o yolo_layer.cu
yolo_layer.h(89): warning: function "nvinfer1::IPluginV2Ext::configurePlugin(const nvinfer1::Dims *, int, const nvinfer1::Dims *, int, const nvinfer1::DataType *, const nvinfer1::DataType *, const __nv_bool *, const __nv_bool *, nvinfer1::PluginFormat, int)" is hidden by "nvinfer1::YoloLayerPlugin::configurePlugin" -- virtual function override intended?

yolo_layer.h(89): warning: function "nvinfer1::IPluginV2Ext::configurePlugin(const nvinfer1::Dims *, int, const nvinfer1::Dims *, int, const nvinfer1::DataType *, const nvinfer1::DataType *, const bool *, const bool *, nvinfer1::PluginFormat, int)" is hidden by "nvinfer1::YoloLayerPlugin::configurePlugin" -- virtual function override intended?

Assembler messages:
Fatal error: can't create yolo_layer.o: Permission denied
Makefile:37: recipe for target 'yolo_layer.o' failed
make: *** [yolo_layer.o] Error 1

Can you give me a hand?

Enso-Zheng, 2021-08-10:
Last time I use x86_64 Pc 
Today I try to use Jeston Agx Xavier get this issue,but Trt still can work!
[image]

akyola, 2021-08-10:
> Last time I use x86_64 Pc 
> Today I try to use Jeston Agx Xavier get this issue,but Trt still can work!
> [image]
> 

I get permission denied error when running make. Yours seems to be successfully generates the output file.

jkjung-avt, 2021-08-11:
@Enso-Zheng  The warning in your screenshot could be saved ignored.  It is due to C++ class declaration weirdness in TensorRT header files.

> Today I try to use Jeston Agx Xavier get this issue,but Trt still can work!

jkjung-avt, 2021-08-11:
```
Assembler messages:
Fatal error: can't create yolo_layer.o: Permission denied
Makefile:37: recipe for target 'yolo_layer.o' failed
make: *** [yolo_layer.o] Error 1
```

@akyola  You should check:
1. permission of the directory,
2. whether there already exists a "yolo_layer.o" which could not be overwritten.  (In this case, you should just delete "yolo_layer.o" and do `make` again.)

Enso-Zheng, 2021-08-16:
> Please check out [README_x86.md](https://github.com/jkjung-avt/tensorrt_demos/blob/master/README_x86.md). You probably need to modify `TENSORRT_INCS` and `TENSORRT_LIBS` in your "plugins/Makefile".

I try to  modify `TENSORRT_INCS` and `TENSORRT_LIBS`  and get this error 
[image]
and did i need to modify 'INCS' and 'LIBS' to my x64PC path or Maintain the status?
thanks~

jkjung-avt, 2021-08-16:
I suspect you are facing the same issue as describe in https://github.com/jkjung-avt/tensorrt_demos/issues/436.

Since you are using a GPU with "compute_86" (Ampere architecture), you need to use [CUDA Toolkit 11.x](https://www.phoronix.com/scan.php?page=news_item&px=NVIDIA-CUDA-11-Released) with a relatively new driver.  In addition, make sure you've installed the matching package of TensorRT 7.2.1.6 for CUDA 11.x.

closed 2021-08-28 by jkjung-avt

aafaqin, 2021-12-28:
If you use the later version of TensorRT make sure in the make file you are writing the correct address.
https://github.com/jkjung-avt/tensorrt_demos/blob/2e31c5deb42cf510030e8f6571b5a49671d70996/plugins/Makefile#L19-L20
```

TENSORRT_INCS=-I"/usr/local/TensorRT-8.2.0.4/include"
TENSORRT_LIBS=-L"/usr/local/TensorRT-8.2.0.4/lib"
```
Can be replaced with a different line  example

Sspacegray, 2022-07-14:
if i use TENSORT5 ,not TENSORRT_INCS and TENSORRT_LIBS,how to modify them?