jkjung-avt/tensorrt_demos #462: meet errors when i try to make in googlenet

Chenhait, 2021-08-16:
[image]
Thank you very much if any advice!

jkjung-avt, 2021-08-17:
Which version of TensorRT are you using?

Chenhait, 2021-08-17:
> Which version of TensorRT are you using?
TensorRT8.0

mtc-20, 2021-08-31:
I ran into a similar problem when trying `build_engine.py` from `ssd` for **JetPack 4.6** and **TensorRT 8**

These changes helped solve the issue for me, might help if you can find the C++ implementations for these API changes

```python
with trt.Builder(TRT_LOGGER) as builder, builder.create_network() as network, \
     builder.create_builder_config() as config, trt.UffParser() as parser:
        config.max_workspace_size = 1 << 28  
        builder.max_batch_size = 1
        config.flags = trt.BuilderFlag.FP16

```

jkjung-avt, 2021-08-31:
I believe the required change (C++ code) is not much.  I'm just struggling to find time to do it...  Will try to look into it as soon as I can...

xuaner11111, 2021-09-08:
> I believe the required change (C++ code) is not much. I'm just struggling to find time to do it... Will try to look into it as soon as I can...

hello, did you solve this problem