jkjung-avt/tensorrt_demos #463: build model with `--dla_core 1` but model looks to run on DLA core 0

y-okumura-isp, 2021-08-17:
In my understanding, we can use DLA core 1 by building the model, but we can not specify the core on runtime.
https://github.com/jkjung-avt/tensorrt_demos/issues/394
Though I set `--dla_core 1` at build time, it looks DLA Core 0 is used.

Here is my environment.

- Board
  - Jetson NX
  - `/etc/nv_tegra_release`: R32 (release), REVISION: 5.1
  - TensorRT 7.1.3.0
- Version
  - 07f3d4af070014177fe5594ec294234ff628dd0f

I built the tensorrt model as following:

```
$ cd yolo
$ python3 onnx_to_tensorrt.py -m yolov4-csp-512 --dla_core 1
Using DLA core 1.
(snip)

$ cd ..
$ python3 trt_yolo.py --image dog.jpg -m yolov4-csp-512
```

And I checked which DLA core is used:
We can do this by checking `/sys/.../runtime_status` according to https://forums.developer.nvidia.com/t/matrix-multiply-on-dla-and-checking-dla-usage/84338/2

```
$ while true; do 
  echo "===";
  cat /sys/devices/platform/host1x/15880000.nvdla0/power/runtime_status;
  cat /sys/devices/platform/host1x/158c0000.nvdla1/power/runtime_status;
  sleep 0.1;
done
```

I got the result as below. It looks DLA Core 0 is used.

```
===
active
suspended
```

All suggestions are welcome.
Thanks.

jkjung-avt, 2021-08-17:
Thanks for reporting the issue.  Below is the relevant source code, and the relevant documentation is here: [IBuilderConfig](https://docs.nvidia.com/deeplearning/tensorrt/api/python_api/infer/Core/BuilderConfig.html#).  I followed NVIDIA's API and set `DLA_Core` in the TensorRT Builder Config.  I'm not sure why it does not work.  I think you could re-direct this question to NVIDIA.

https://github.com/jkjung-avt/tensorrt_demos/blob/9dd56b3b8d841dcfc2e5d1868f4bd785a50cbe98/yolo/onnx_to_tensorrt.py#L159-L164

y-okumura-isp, 2021-08-18:
Thank you for your comment.
It looks the code is correct.
I try to post NVIDIA Forum.
By the way, can anyone reproduce this situation? How about on boards except for NX?

closed 2021-08-28 by jkjung-avt

xiaochus, 2021-12-08:
I have the same question and find the solution on my device . I try config->setDLACore(0) and config->setDLACore(1), but the model always run on DLA 0. Then,  i try setDLACore at runtime and the model can run on the correct DLA.