jkjung-avt/tensorrt_demos #466: can we do setup in google collab?? want to convert yolov4x-mish --> yolov4x-mish.onnx -->> yolov4x.trt

akashAD98, 2021-08-19:
im following the your guidance for setting up enviromment in google collab

`!git clone https://github.com/jkjung-avt/tensorrt_demos.git`

`cd tensorrt_demos`

`cd /content/tensorrt_demos/ssd`
`! ./install.sh`
`!pip install onnx==1.4.1`
`cd /content/tensorrt_demos/plugins`
`!make``

Im getting 
this error 
##
computes: 75
NVCCFLAGS: -gencode arch=compute_75,code=[sm_75,compute_75]
nvcc --std c++11 -ccbin g++ -I"/usr/local/cuda/include" -I"/usr/local/TensorRT-7.1.3.4/include" -I"/usr/local/include" -I"plugin" -gencode arch=compute_75,code=[sm_75,compute_75] -Xcompiler "-fPIC" -c -o yolo_layer.o yolo_layer.cu
In file included from yolo_layer.cu:14:0:
yolo_layer.h:9:10: fatal error: NvInfer.h: No such file or directory
 #include "NvInfer.h"
          ^~~~~~~~~~~
compilation terminated.
Makefile:37: recipe for target 'yolo_layer.o' failed
make: *** [yolo_layer.o] Error 1

i check this isse https://github.com/jkjung-avt/tensorrt_demos/issues/459
i did respective changes, still getting the error.

##

jkjung-avt, 2021-08-19:
I don't think TensorRT libraries are available in a Colab runtime by default.  So you'll need to install TensorRT by yourself.  I'm not sure if it's doable, though.

closed 2021-08-28 by jkjung-avt

Kanyade, 2021-09-02:
@akashAD98 
Here is my solution:

https://gist.github.com/Kanyade/c8ee3b7eca6f6e470e418b018ea02e83

It is for yolov3 but I think you can benefit from it

Special thanks to @prratadiya for this repo: https://github.com/prratadiya/tensorrt-installation-colab
Basically it's the same as his, I just added the required steps for the tensorrt_demos project as well