jkjung-avt/tensorrt_demos #467: Yolov5

abhigoku10, 2021-08-19:
@jkjung-avt  thanks for the code base do you have onnx_to_tensorrt.py for yolov5 ??

jkjung-avt, 2021-08-19:
I have not tried it myself.  But if you manage to convert a YOLOv5 model to ONNX, "onnx_to_tensorrt.py" would likely work.

abhigoku10, 2021-08-19:
@jkjung-avt  thanks for the response , will this work any version of the yolo since i see that in #56 your getting data from Darknetparser "from yolo_to_onnx import DarkNetParser, get_h_and_w"

jkjung-avt, 2021-08-19:
Right.  My code was designed for DarkNet models.  If you are not using a DarkNet model, you'll have to modify the source code...

abhigoku10, 2021-08-19:
@jkjung-avt  thanks for it !!! any future plans to make more independent of darknet module !!!

jkjung-avt, 2021-08-20:
> any future plans to make more independent of darknet module !!!

I don't have such a plan now.

abhigoku10, 2021-08-20:
@jkjung-avt  thankf or response

closed 2021-08-20 by abhigoku10