jkjung-avt/tensorrt_demos #469: a little question for plugin yolo_layer.

credoxfromcqut, 2021-08-20:
hi, contributor. i cant find that where does the libyolo_layer.so been used. i checked the plugin.py and yolo_with_plugin.py , they just try to load the .so files and didnt assign to an object，how does this process (yolo_layer.so) going...

jkjung-avt, 2021-08-20:
The `YoloPluginCreator` registers itself to TensorRT library with the name ["YoloLayer_TRT"](https://github.com/jkjung-avt/tensorrt_demos/blob/1f8773b3f6091d2e3441c0d6ce5714e332421f8d/plugins/yolo_layer.cu#L280).

https://github.com/jkjung-avt/tensorrt_demos/blob/1f8773b3f6091d2e3441c0d6ce5714e332421f8d/plugins/yolo_layer.h#L147

So when TensorRT library encounters a "YoloLayer_TRT" layer in the engine, it knows it should delegate the handling to `YoloPluginCreator` which creates the `YoloLayerPlugin` object.  All those are implemented in "libyolo_layer.so".

https://github.com/jkjung-avt/tensorrt_demos/blob/1f8773b3f6091d2e3441c0d6ce5714e332421f8d/utils/yolo_with_plugins.py#L18

credoxfromcqut, 2021-08-20:
great THANKS to you !

closed 2021-08-20 by credoxfromcqut