jkjung-avt/tensorrt_demos #471: Support for JetPack 4.6

Dreadlocke1984, 2021-08-24:
Hi,
    We currently installed JetPack 4.6 on our Jetson Xavier AGX. It has TensorRT 8 preinstalled. However, we are unable to compile the create_engines file. From our understanding the last TensorRT version tested was version 5.

Are there some additional steps that are required for TensorRT 8? We would really appreciate the help.

jkjung-avt, 2021-08-24:
I think this is the same issue as https://github.com/jkjung-avt/tensorrt_demos/issues/462.

The problem is due to API change between TensorRT 8 and TensorRT 7.  I'll need to find time to update the code...

ismet-dagli, 2021-09-09:
Just to confirm, is it related to support on jetpack 4.6 as well?

Command to run: python3 yolo_to_onnx.py -m yolov4-416

> Checking ONNX model...
> Traceback (most recent call last):
>   File "yolo_to_onnx.py", line 983, in <module>
>     main()
>   File "yolo_to_onnx.py", line 974, in main
>     onnx.checker.check_model(yolo_model_def)
>   File "/home/xaviernx/.local/lib/python3.6/site-packages/onnx/checker.py", line 93, in check_model
>     C.check_model(model.SerializeToString())
> onnx.onnx_cpp2py_export.checker.ValidationError: Op registered for Upsample is deprecated in domain_version of 12
> 
> ==> Context: Bad node spec: input: "118_convolutional_lrelu" input: "119_upsample_scale" output: "119_upsample" name: "119_upsample" op_type: "Upsample" attribute { name: "mode" s: "nearest" type: STRING }

jkjung-avt, 2021-09-09:
> Command to run: python3 yolo_to_onnx.py -m yolov4-416

"yolo_to_onnx.py" and "onnx_to_tensorrt.py" should not be affected, since I used python API in those and already accounted for TensorRT API changes.  I think they should just work on a JetPack-4.6 system.

The thing I need to fix is for trt_googlenet and trt_mtcnn.  They were implemented with C++ API.  I still need to allocate some time to do it.