jkjung-avt/tensorrt_demos #475: python3: yolo_layer.cu:121: virtual nvinfer1::Dims nvinfer1::YoloLayerPlugin::getOutputDimensions(int, const nvinfer1::Dims*, int): Assertion `inputs[0].d[1] == mYoloHeight' failed. Aborted (core dumped)

jaabaal, 2021-09-01:
Hi @jkjung-avt sir! I have modified the configuration file of the yolov4-tiny.cfg and I am trying to use the onnx_to_trt.py script to serialize my .onnx model to .trt. However I am getting the above specified error. Can you please help me with that? I am attaching my config file for your reference as well.
[yolov4-tiny-vgg-16.cfg.zip](https://github.com/jkjung-avt/tensorrt_demos/files/7093440/yolov4-tiny-vgg-16.cfg.zip)

jkjung-avt, 2021-09-02:
The "yolov4-tiny-vgg-16" model you're using has a different architecture from the original "yolov4-tiny" model.  It does not contain FPN, i.e. no "upsample" layer.

You'll have to modify the source code by yourself to get it to work.

Here are the code pieces that you'd modify:

1. https://github.com/jkjung-avt/tensorrt_demos/blob/1f8773b3f6091d2e3441c0d6ce5714e332421f8d/yolo/yolo_to_onnx.py#L981-L982

   Replace the above with the following:

   ```
       output_tensor_shapes = [
           [c, h // 16, w // 16], [c, h // 32, w // 32]] 
   ```

2. https://github.com/jkjung-avt/tensorrt_demos/blob/1f8773b3f6091d2e3441c0d6ce5714e332421f8d/yolo/plugins.py#L91-L92

   Replace the above with the following:

   ```
           yolo_whs = [
               [w // 16, h // 16], [w // 32, h // 32]]
   ```

jaabaal, 2021-09-02:
Hello @jkjung-avt Sir. The following changes worked and my code is working. Thank you very much for your help! I had made the changes in the yolo_to_onnx.py file but did not know about the one in plugins.py. Once again, thank you very much. :)

closed 2021-09-02 by jaabaal