jkjung-avt/tensorrt_demos #476: onnx to trt failed.. "cannot get YoloLayer_TRT plugin creator"

seungtaek94, 2021-09-03:
Hi. I try to convert yolo to trt. 

I succeed to convert .cfg -> .onnx but failed .onnx -> .trt with Error "cannot get YoloLayer_TRT plugin creator".

What should I check now and what does mean that error message.?

I already `make` in plugins dir and get `libyolo_layer.so` lib.

jkjung-avt, 2021-09-03:
Please provide the complete log of "onnx_to_tensorrt.py".

seungtaek94, 2021-09-05:
@jkjung-avt 

```
(trt) seungtaek@seungtaek:~/trt/tensorrt_demos/yolo$ python onnx_to_tensorrt.py -m yolov4-p5-640x384
Loading the ONNX file...
Adding yolo_layer plugins.
Traceback (most recent call last):
  File "onnx_to_tensorrt.py", line 212, in <module>
    main()
  File "onnx_to_tensorrt.py", line 201, in main
    args.model, args.int8, args.dla_core, args.verbose)
  File "onnx_to_tensorrt.py", line 123, in build_engine
    network = add_yolo_plugins(network, model_name, TRT_LOGGER)
  File "/home/seungtaek/trt/tensorrt_demos/yolo/plugins.py", line 123, in add_yolo_plugins
    raise RuntimeError('cannot get YoloLayer_TRT plugin creator')
RuntimeError: cannot get YoloLayer_TRT plugin creator
```

Here :)

jkjung-avt, 2021-09-06:
You are not using the latest code...  Please `git pull` the latest code, rebuild the "yolo_layer" plugin (`make clean; make`), and re-run "yolo_to_onnx.py" and "onnx_to_tensorrt.py".

The `raise RuntimeError(...)` is on line 122 (not 123) in the latest code.

https://github.com/jkjung-avt/tensorrt_demos/blob/1f8773b3f6091d2e3441c0d6ce5714e332421f8d/yolo/plugins.py#L121-L122

seungtaek94, 2021-09-06:
@jkjung-avt 

```
def get_plugin_creator(plugin_name, logger):
    """Get the TensorRT plugin creator."""
    trt.init_libnvinfer_plugins(logger, '')
    plugin_creator_list = trt.get_plugin_registry().plugin_creator_list
    for c in plugin_creator_list:
        #print(c.name)   # << here
        if c.name == plugin_name:
            return c
    return None
```

No, Im using latest version. I added only one line that `print(c.name)` in `plugins.py`. So that is why line number is different with latest code.

jkjung-avt, 2021-09-06:
If this `ctypes.cdll.LoadLibrary('../plugins/libyolo_layer.so')` succeeds, "YoloLayer_TRT" plugin creator should have been registered into the TensorRT...

https://github.com/jkjung-avt/tensorrt_demos/blob/1f8773b3f6091d2e3441c0d6ce5714e332421f8d/yolo/plugins.py#L16-L17

Do you see something similar when you do `print(c.name)` on all TensorRT plugin creators?

Otherwise, which version of TensorRT are you using?

seungtaek94, 2021-09-06:
@jkjung-avt 

I use `TensorRT-7.1.3.4`

Below is `print(c.name)`  output.. `YoloLayer_TRT` is not registered.. But i dont know why.
```
CustomQKVToContextPluginDynamic
CustomEmbLayerNormPluginDynamic
RnRes2Br1Br2c_TRT
RnRes2Br1Br2c_TRT
CustomSkipLayerNormPluginDynamic
CustomSkipLayerNormPluginDynamic
CustomSkipLayerNormPluginDynamic
GroupNormalizationPlugin
CustomGeluPluginDynamic
CustomEmbLayerNormPluginDynamic
CgPersistentLSTMPlugin_TRT
CustomQKVToContextPluginDynamic
CustomQKVToContextPluginDynamic
CustomFCPluginDynamic
SingleStepLSTMPlugin
RnRes2Br2bBr2c_TRT
RnRes2Br2bBr2c_TRT
RnRes2FullFusion_TRT
SmallTileGEMM_TRT
RNNTEncoderPlugin
GridAnchor_TRT
NMS_TRT
Reorg_TRT
Region_TRT
Clip_TRT
LReLU_TRT
PriorBox_TRT
Normalize_TRT
RPROI_TRT
BatchedNMS_TRT
BatchedNMSDynamic_TRT
FlattenConcat_TRT
CropAndResize
DetectionLayer_TRT
Proposal
ProposalLayer_TRT
PyramidROIAlign_TRT
ResizeNearest_TRT
Split
SpecialSlice_TRT
InstanceNormalization_TRT
```

jkjung-avt, 2021-09-08:
Beats me...

Did you see any errors/warnings when building the plugin?

seungtaek94, 2021-09-09:
@jkjung-avt 

Below is all outputs when i do `make` in plugins forlder.

```
computes: 61
NVCCFLAGS: -gencode arch=compute_61,code=[sm_61,compute_61]
nvcc -ccbin g++ -I"/usr/local/cuda-11.1/include" -I"/usr/local/TensorRT-7.1.3.4/include" -I"/usr/local/include" -I"plugin" -gencode arch=compute_61,code=[sm_61,compute_61] -Xcompiler -fPIC -c -o yolo_layer.o yolo_layer.cu
yolo_layer.h(89): warning: function "nvinfer1::IPluginV2Ext::configurePlugin(const nvinfer1::Dims *, int32_t, const nvinfer1::Dims *, int32_t, const nvinfer1::DataType *, const nvinfer1::DataType *, const __nv_bool *, const __nv_bool *, nvinfer1::PluginFormat, int32_t)" is hidden by "nvinfer1::YoloLayerPlugin::configurePlugin" -- virtual function override intended?

yolo_layer.h(89): warning: function "nvinfer1::IPluginV2Ext::configurePlugin(const nvinfer1::Dims *, int32_t, const nvinfer1::Dims *, int32_t, const nvinfer1::DataType *, const nvinfer1::DataType *, const bool *, const bool *, nvinfer1::PluginFormat, int32_t)" is hidden by "nvinfer1::YoloLayerPlugin::configurePlugin" -- virtual function override intended?

g++ -shared -o libyolo_layer.so yolo_layer.o -L"/usr/local/cuda-11.1/lib64" -L"/usr/local/TensorRT-7.1.3.4/lib" -L"/usr/local/lib" -Wl,--start-group -lnvinfer -lnvparsers -lnvinfer_plugin -lcudnn -lcublas -lnvToolsExt -lcudart -lrt -ldl -lpthread -Wl,--end-group
```

jkjung-avt, 2021-09-09:
> Below is all outputs when i do make in plugins folder.

The log looks completely normal.  (The virtual function override warnings could be safely ignored.)  I don't really have on idea why you met this problem ("cannot get YoloLayer_TRT plugin creator")...

jkjung-avt, 2021-09-24:
I'm closing this issue since I myself am not able to reproduce the problem.

closed 2021-09-24 by jkjung-avt

sebawsky, 2021-10-06:
Hello there,

I had exactly the same error, I could not get the script "onnx_to_tensorrt.py -m yolov4-416" to work. Has anyone found the solution?

sebawsky, 2021-10-06:
@jkjung-avt Yes, I ran as it appears in the tutorial. I followed the tutorial on my Jetson Nano and I was able to make the network work without problems but in ubuntu18 I could not (I still have this error).

Greetings

akashAD98, 2021-10-07:
Im also getting the same issue 

 File "onnx_to_tensorrt.py", line 212, in <module>
    main()
  File "onnx_to_tensorrt.py", line 200, in main
    engine = build_engine(
  File "onnx_to_tensorrt.py", line 123, in build_engine
    network = add_yolo_plugins(network, model_name, TRT_LOGGER)
  File "/home/ubuntu/tensorrt_demos/yolo/plugins.py", line 122, in add_yolo_plugins
    raise RuntimeError('cannot get YoloLayer_TRT plugin creator')

 cannot get YoloLayer_TRT plugin creator

i have sucesfully converte3d yolo yo onnx , but getting the exactly same issue,
Even i checked path of libyolo_layer.so
& added here still getting the same issue 

try:
    ctypes.cdll.LoadLibrary('/home/ubuntu/tensorrt_demos/plugins/libyolo_layer.so')
   
Your help is really appreciated.Thanks @jkjung-avt

akashAD98, 2021-10-08:
@jkjung-avt my server is off , once its switched on ill do the testing

jkjung-avt, 2021-10-09:
@sebawsky @akashAD98  I see.  So the problem is not loading of the dynamic library ("libyolo_layer.so.so").  Rather, `REGISTER_TENSORRT_PLUGIN()` does not work as expected ("YoloLayer_TRT" plugin creator is not registered).

The relevant source code is here.  I don't know why, but will investigate more when I have time.

https://github.com/jkjung-avt/tensorrt_demos/blob/ca6ac81b4c7558122df3ff3018479412828a88a4/plugins/yolo_layer.h#L147

akashAD98, 2021-10-09:
@jkjung-avt is there any temporary solution for this? it's a little bit urgent for me, Should we try a new setup? new machine? any advice will be really helpful

akashAD98, 2021-10-09:
@jkjung-avt @sebawsky I solved this issue by uninstalling the old tensorrt version.pycuda & onxx 
& i did again setup with TensorRT-8.0.3.4 ,& its working fine . & again thanks for your valuable time & reply ):

jkjung-avt, 2021-10-12:
@akashAD98  Please remove the whole "src/pycuda-2019.1.2" directory and re-run "install_pycuda.sh".

```shell
$ rm -rf ${HOME}/src/pycuda-2019.1.2/
$ cd ${HOME}/project/tensorrt_demos/ssd/
$ ./install_pycuda.sh
```

PuneethBC, 2021-12-10:
I had the same issue of "RuntimeError: cannot get YoloLayer_TRT plugin creator", because the environment CUDA_VISIBLE_DEVICES was set to -1.
after export CUDA_VISIBLE_DEVICES=0, it worked.

sachinkmohan, 2022-05-03:
> @sebawsky @akashAD98 I see. So the problem is not loading of the dynamic library ("libyolo_layer.so.so"). Rather, `REGISTER_TENSORRT_PLUGIN()` does not work as expected ("YoloLayer_TRT" plugin [...]

@jkjung-avt , I am still confused. I can't go any further than this. Did you happen to check this?