jkjung-avt/tensorrt_demos #477: [Free gpu memory when not using]

kiepluhanh97, 2021-09-08:
Hi,
- I run your yolo inference code in yolo_with_plugins.py file. Everything ok but I have a trouble with multithread running.

- I can load many model run on multi thread. But when I stop one of each. My memory in gpu was not free even I checked function __dell__() called in TrtYOLO class. 

- Can you tell me how can I manage and free all gpu memory in __dell__() funtion.?

- I realized my gpu memory used more in  "trt.Runtime.deserialize_cuda_engine(),  create_execution_context(), allocate_buffers() " functions when I start new thread and loaded new model in that thread.

-  My code look like below:

yoloWeightPath = "yolo_model_path"
trt_yolo1 = TrtYOLO(yoloWeightPath, (416, 416), 1)
trt_yolo2 = TrtYOLO(yoloWeightPath, (416, 416), 1) # need more GPU memory
trt_yolo2.__dell__() # called when stop thread using that. GPU memory not be free?

jkjung-avt, 2021-09-17:
Replace this:

   ```
   trt_yolo2.del()
   ```

with the following and see if it makes a difference.

   ```
   del(trt_yolo2)
   ```

Otherwise, check out https://github.com/NVIDIA/TensorRT/issues/582 to see if it helps.

jkjung-avt, 2021-09-24:
Closing due to no response.

closed 2021-09-24 by jkjung-avt