jkjung-avt/tensorrt_demos #478: Issue with rtsp stream

basitraza1214, 2021-09-15:
I am facing the following error while accessing the RTSP stream from a camera that supports the RTSP stream. I am able to access the stream in vlc as well as in the shell using GStreamer.

This is my input:

> python3 trt_yolo.py --rtsp "rtsp://admin:admin@123@192.168.1.108:554/cam/realmonitor?channel=1&subtype=0" --m yolov4-custom-416 --category_num 1

This the shell output:

> [ WARN:0] global /home/nvidia/host/build_opencv/nv_opencv/modules/videoio/src/cap_gstreamer.cpp (711) open OpenCV | GStreamer warning: Error opening bin: syntax error
[ WARN:0] global /home/nvidia/host/build_opencv/nv_opencv/modules/videoio/src/cap_gstreamer.cpp (480) isPipelinePlaying OpenCV | GStreamer warning: GStreamer: pipeline have not been created
WARNING:root:Camera: starting while cap is not opened!
ERROR: failed to open camera!

jkjung-avt, 2021-09-17:
Reference:

* [How to Capture and Display Camera Video with Python on Jetson TX2](https://jkjung-avt.github.io/tx2-camera-with-python/)
* https://github.com/jkjung-avt/tensorrt_demos/issues/295

And at least make sure:

* You have installed all necessary GStreamer plugins for receiving/decoding RTSP streams.
* Check whether your RTSP stream is encoded in H.264 or H.265.  Modify the source code for H.265 if necessary.

jkjung-avt, 2021-09-24:
Closing due to no further response.

closed 2021-09-24 by jkjung-avt