jkjung-avt/tensorrt_demos #480: GStreamer issue: using an HTTP streaming as input

BenjaminQU259, 2021-09-29:
Hi,

I was working with [Demo #5: YOLOv4](https://github.com/jkjung-avt/tensorrt_demos#yolov4) on my Jetson AGX Xavier, and it worked flawlessly when I used an MP4 video file as input.

But now, I need to use HTTP streaming as the input. So I tried the `open_cam_gstr`, and passed the argument as:
`python3 trt_yolo.py --gstr 'souphttpsrc location=http://{username}:{passwd}@{IP_address}:{port}/stream.mjpg ! jpegdec ! xvimagesink' -m {model}`

Then I got this error:
`[ WARN:0] global /home/nvidia/host/build_opencv/nv_opencv/modules/videoio/src/cap_gstreamer.cpp (801) open OpenCV | GStreamer warning: cannot find appsink in manual pipeline`
`[ WARN:0] global /home/nvidia/host/build_opencv/nv_opencv/modules/videoio/src/cap_gstreamer.cpp (480) isPipelinePlaying OpenCV | GStreamer warning: GStreamer: pipeline have not been created`
`WARNING:root:Camera: starting while cap is not opened!`
`ERROR: failed to open camera!`

I'm not sure how this error occurred, but I went ahead and tried opening the streaming directly with GStreamer:
`gst-launch-1.0 souphttpsrc location=http://{username}:{passwd}@{IP_address}:{port}/stream.mjpg ! jpegdec ! xvimagesink`,
and it worked. I'm sure the steaming address is correct.

I'm new to this and I don't know where the problem is.

Any help would be appreciated.

jkjung-avt, 2021-09-29:
Try this instead:

> python3 trt_yolo.py --gstr 'souphttpsrc location=http://{username}:{passwd}@{IP_address}:{port}/stream.mjpg ! videoconvert ! appsink' -m {model}

BenjaminQU259, 2021-09-29:
Thank you for the quick reply. So I tried:
> python3 trt_yolo.py --gstr 'souphttpsrc location=http://{username}:{passwd}@{IP_address}:{port}/stream.mjpg ! videoconvert ! appsink' -m {model}

And now I got this:

<pre>[ WARN:0] global /home/nvidia/host/build_opencv/nv_opencv/modules/videoio/src/cap_gstreamer.cpp (1757) handleMessage OpenCV | GStreamer warning: Embedded video playback halted; module souphttpsrc0 reported: Internal data stream error.
[ WARN:0] global /home/nvidia/host/build_opencv/nv_opencv/modules/videoio/src/cap_gstreamer.cpp (886) open OpenCV | GStreamer warning: unable to start pipeline
[ WARN:0] global /home/nvidia/host/build_opencv/nv_opencv/modules/videoio/src/cap_gstreamer.cpp (480) isPipelinePlaying OpenCV | GStreamer warning: GStreamer: pipeline have not been created
WARNING:root:Camera: starting while cap is not opened!
ERROR: failed to open camera!</pre>

jkjung-avt, 2021-09-29:
Sorry, my bad.  I missed the jpeg decoder in the pipeline.

> python3 trt_yolo.py --gstr 'souphttpsrc location=http://{username}:{passwd}@{IP_address}:{port}/stream.mjpg ! jpegdec ! videoconvert ! appsink' -m {model}

BenjaminQU259, 2021-09-29:
OK it worked:
> python3 trt_yolo.py --gstr 'souphttpsrc location=http://{username}:{passwd}@{IP_address}:{port}/stream.mjpg ! jpegdec ! videoconvert ! appsink' -m {model}

Fantastic! Thank you very much!

closed 2021-09-29 by BenjaminQU259