jkjung-avt/tensorrt_demos #481: Tensorrt RTSP Server

seabass1217, 2021-09-30:
@jkjung-avt,

First off thank you for your work!  I wish I could give you 10 stars!  I have more of a question than an issue.  Are you aware of a python-based rtsp server that easily integrates with your code (trt_yolo.py) like the python mjpeg-sever does (in trt_yolo_mjpeg.py)?  

I would like to stream yolo detections like the trt_yolo_mjpeg.py script does except with an rtsp server instead of a mjpeg sever.

Any recommendations are greatly appreciated.

Thanks!

jkjung-avt, 2021-10-02:
Thanks for your endorsement.

I don't currently have a plan to add RTSP server support to the code (I welcome such a pull request, though).  You could probably refer to how I create a VideoWriter with a GStreamer pipeline in the trt_modnet example.

https://github.com/jkjung-avt/tensorrt_demos/blob/1f8773b3f6091d2e3441c0d6ce5714e332421f8d/utils/writer.py#L20-L24

https://github.com/jkjung-avt/tensorrt_demos/blob/1f8773b3f6091d2e3441c0d6ce5714e332421f8d/trt_modnet.py#L154-L155

https://github.com/jkjung-avt/tensorrt_demos/blob/1f8773b3f6091d2e3441c0d6ce5714e332421f8d/trt_modnet.py#L133

https://github.com/jkjung-avt/tensorrt_demos/blob/1f8773b3f6091d2e3441c0d6ce5714e332421f8d/trt_modnet.py#L164-L165

And try to replace the `filesink` with something that produces a RTSP stream.  Reference:

* https://forums.developer.nvidia.com/t/how-to-generate-rtsp-stream-with-gstreamer-in-tx2/75876
* https://stackoverflow.com/questions/13744560/using-gstreamer-to-serve-rtsp-stream-working-example-sought

seabass1217, 2021-10-02:
@jkjung-avt, Thanks for the reply!  I'm sure I will find a solution from your recommendations.  I consider this issue closed.

closed 2021-10-02 by seabass1217