jkjung-avt/tensorrt_demos #482: is there any planning for supporting  yolo9000 to tensorrt?

akashAD98, 2021-10-01:
this is repo : https://github.com/philipperemy/yolo-9000
https://github.com/mathieuorhan/darknet/blob/master/README.md#using-yolo9000,
Thank you so much for your amazing work, please let us know if there is any planning for this yolo9000 support,

jkjung-avt, 2021-10-02:
I think my code would work for DarkNet YOLOv2 directly.  Feel free to give it a try.

akashAD98, 2021-10-04:
@jkjung-avt should I try this demo yolov3,yolov4?

jkjung-avt, 2021-10-04:
> should I try script of yolov3,yolov4 for supporting yolo-9000?

Yes.

jkjung-avt, 2021-10-12:
@akashAD98  The "region" layer seems to have been replaced by the "yolo" layer in YOLOv3 and YOLOv4.  This repo does not support it.  Sorry, I was not aware that YOLO9000 would require support of the "region" layer.

It would require some code modification to make it work.  I don't have a plan to do that...