jkjung-avt/tensorrt_demos #485: Low MODNet output feature similarity between trtexec and onnx

jackgao0323, 2021-10-08:
Hi,
I'm comparing the MODNet output feature between  onnx and trtexec with zeros input and getting the cosine similarity only 0.00029. Do anyone have any idea of this issue? Btw, the result looks great.

jkjung-avt, 2021-10-12:
Could you elaborate the question?

How (what steps) do you check the problem?  What results do you expect?

jkjung-avt, 2021-10-20:
For the `Cosine similarity between ONNX and TensorRT outputs: 0.9999999` result, I did the testing on Jetson Nano with JetPack-4.6.  It is CUDA 10.2, cuDNN 8.2.1 and TensorRT 8.0.1.6.

jkjung-avt, 2021-10-20:
Sorry about my typo in the previous post.

The values in your picture are all `-0.5`.  Those are not valid output values of MODNet.

jackgao0323, 2021-10-20:
> Sorry about my typo in the previous post.
> 
> The values in your picture are all `-0.5`. Those are not valid output values of MODNet.

Yes, it's pretty weird.🤔🤔
I'll cross validation on my Jetson NX at the same time.

closed 2021-11-01 by jackgao0323