jkjung-avt/tensorrt_demos #499: yolov3-tiny-416 onnx2trt error

Wooks-git, 2021-11-08:
Hi, when i follow your command for convert yolov3-tiny-416 model to .trt , i have this error.

[TensorRT] ERROR: ../rtSafe/safeRuntime.cpp (25) - Cuda Error in allocate: 999 (unknown error)

So i tried https://forums.developer.nvidia.com/t/error-trt-rtsafe-saferuntime-cpp-25-cuda-error-in-allocate-999-unknown-error/174814 solution, but still have same problem.
How can i fix it?

jkjung-avt, 2021-11-08:
https://docs.nvidia.com/cuda/cuda-runtime-api/group__CUDART__TYPES.html#group__CUDART__TYPES_1g3f51e3575c2178246db0a94a430e0038

> cudaErrorUnknown = 999
> This indicates that an unknown internal error has occurred.

This might be related to your NVIDIA driver and library (CUDA, cuDNN, TensorRT, etc.) installations.  I think you need to check your environment.

Wooks-git, 2021-11-08:
my env
jetpack : 4.5.1
pycuda : 2020.1
onnx : 1.4.1
torch : 1.6.0
,,,

closed 2021-11-10 by Wooks-git