jkjung-avt/tensorrt_demos #501: yolov4-csp training with custom-dataset

Wooks-git, 2021-11-11:
Hi, could you share the link that you trained yolov4-csp?
I want to train yolov4-csp256 with my custom dataset, but I don't know exactly if this link is right.

https://github.com/WongKinYiu/ScaledYOLOv4/tree/yolov4-csp

jkjung-avt, 2021-11-11:
I only used the pre-trained DarkNet yolov4-csp: https://github.com/AlexeyAB/darknet/blob/master/cfg/yolov4-csp.cfg

Wooks-git, 2021-11-11:
Then, if I want to learn yolov4-csp-256, do I only need to adjust the width height?

jkjung-avt, 2021-11-11:
> Then, if I want to learn yolov4-csp-256, do I only need to adjust the width height?

Yes, I think so.

Wooks-git, 2021-11-17:
Hi Thank you for reply!

I finished my yolov4-csp-256 model. How do i modify yolo_to_onnx.py and onnx_to_trt.py?

trained classes : 3 (face, phone, cigar)

jkjung-avt, 2021-11-17:
1. "yolo_to_onnx.py" and "onnx_to_trt.py" should just work.  No need to modify them.
2. Refer to https://github.com/jkjung-avt/tensorrt_demos/issues/421 for how to modify the code to display your own custom class names at inference time.

Wooks-git, 2021-11-17:
ok thanks a lot!
i'll try it

closed 2021-11-24 by jkjung-avt