jkjung-avt/tensorrt_demos #502: Stream Inference Output

seabass1217, 2021-11-14:
Hi,

I would like to stream the inference output so users can access them via a web-browser.  Similar to  AlexeyAB /
darknet (https://github.com/AlexeyAB/darknet#how-to-use-on-the-command-line) project where users can access predictions in JSON format via a web-browser.

Is there a simple python HTTP/JSON server (or something of the like) that allows inference output to be streamed and access with a browser?
The  stream can be in any format, JSON, XML, or whatever.

Any help or suggested solutions are greatly appreciated.

Thanks!

jkjung-avt, 2021-11-14:
Yes, I have done something similar in this repo.  The [trt_yolo_mjpeg.py](https://github.com/jkjung-avt/tensorrt_demos/blob/master/trt_yolo_mjpeg.py) uses a HTTP server to stream MJPEG images to a remote client.

You could refer to the HTTP server implementation in [utils/mjpeg.py](https://github.com/jkjung-avt/tensorrt_demos/blob/master/utils/mjpeg.py).  It should not be too difficult to extend the `MjpegServer` and `MjpegHandler` classes to support `GET` of JSON detection results.

seabass1217, 2021-11-15:
@jkjung-avt,

Thank you for your feedback!   I'm fairly new to HTTP/MJPEG Server but will research your suggestion.  Any reference implementation would be helpful.

Again thank you and I will close this issue.

closed 2021-11-15 by seabass1217