jkjung-avt/tensorrt_demos #504: fp32 vs FP16 vs int8 , both fp32 & fp16 having same results & fps?

akashAD98, 2021-11-16:
i have seen your implementation,its  by default supports fp16, & I tested the results on frames, there are lots of FP, so what i thout if i used FP32 (which is used in darknet) then there will be improvment in accuracuy.
But FPS & accuracy of both fp16 & fp32 is same. 

i turned off fp16 using as per your suggetions  https://github.com/jkjung-avt/tensorrt_demos/issues/450
 builder.fp16_mode =   False & converted yolo.onnx into yolo.trt , 
 & both speed (FPS)  & results are almost same.  
 
i think its not using fp32 . can you explain me how to use fp32

jkjung-avt, 2021-11-17:
If you are using TensorRT 7+, you'll need to comment out (or delete) this line of code instead.

https://github.com/jkjung-avt/tensorrt_demos/blob/fb2f075cc44a344062b31fa924bb9c4595ec7de4/yolo/onnx_to_tensorrt.py#L137

akashAD98, 2021-11-17:
yes it's working fine. but no effect on accuracy, still it's the same. I thought due to FP32 little bit of accuracy will improve.

jkjung-avt, 2021-11-17:
The accuracy issue (mAP drop for TensorRT YOLO engines) has been summarized/tracked in https://github.com/jkjung-avt/tensorrt_demos/issues/315.

closed 2021-11-18 by akashAD98