jkjung-avt/tensorrt_demos #505: Support for newer version of onnx for YOLO-v4

DMLON, 2021-11-16:
Hello! 
We are trying to build the YOLO v4 model into tensorrt in a RTX 3070. The main issue we are having is that it needs at least TensorRT 7.2 but when installing onnx 1.4.1 says that needs an older version of protobuf compiler.

We are evaluation the possibility of upgrading onnx to a newer version but as you stated on the readme, it needs specifically 1.4.1. Do you have any ideas on how to face the issues of upgrading?

Thanks

jkjung-avt, 2021-11-17:
The requirement of "onnx==1.4.1" (opset 9) is for implementation of the "Upsample" layers in the DarkNet models.  Refer to https://github.com/jkjung-avt/tensorrt_demos/issues/202 for more details.

I think there are at least 2 ways that you could solve this problem:

1. Create a python virtual environment with "onnx==1.4.1", and only use this virtualenv to run "yolo_to_onnx.py".  Once you have the ONNX file, you could switch back to your normal environment to build and run the TensorRT engine.

2. Update the source code to support "Upsample" layers with newer ONNX opset.  The most relevant code is here:

   https://github.com/jkjung-avt/tensorrt_demos/blob/f53b5ae9b004489463a407d8e9b230f39230d051/yolo/yolo_to_onnx.py#L886

DMLON, 2021-11-18:
Excelent! The first option worked like a charm. Thanks a lot! Will check the second for later.

closed 2021-11-18 by DMLON

jkjung-avt, 2022-05-06:
,Refer to: https://github.com/jkjung-avt/tensorrt_demos/issues/539#issuecomment-1119218462

I've updated the code to use onnx=1.9.0 now.