jkjung-avt/tensorrt_demos #506: error occurred when ./build_engines.sh with custom model (ssd_mobilenet_v1_coco)

goeunsong, 2021-11-17:
hi,
I customized object detection model using sd_mobilenet_v1_coco (Tensorflow version 1.15) along the tensorflow-object-detection-api-tutorial. Then I created my_ssd_mobilenet_v1_coco.pb and changed build_engine.py  build_engine.sh for my customed model. The problem occurred when I ran ./build_engines.sh. When I ran ./build_engines.sh the file format .pbtxt and .uff was created but .bin has not been created.

The error msg was,
"#assertionnmsPlugin.cpp,249
./build_engines.sh: line 12: 12300 Aborted
(core dumped) python3 build_engine.py my_ssd_mobilenet_v1_coco".

There's problem when executing engine = builder.build_cuda_engine(network) in build_engine.py. I'd like to know if there's anything I'm missing.
Thanks.

closed 2021-11-17 by goeunsong

JimmyHu1107, 2021-12-20:
and how you solved the probelm? I met the same error .thanks.

goeunsong, 2021-12-21:
My problem was that i got wrong num_classes. As the number of classes had to include a background, so it had to be one more value than the number of classes.