jkjung-avt/tensorrt_demos #509: Try to Convert onnx to tensorrt int8  but got Error

3a532028, 2021-11-24:
I try to convert onnx to tensorrt int8  but got this error bellow,would some one help me up?

$ python3 onnx_to_tensorrt.py -m yolov4-416 --int8

~~~

Loading the ONNX file...
Adding yolo_layer plugins.
Adding a concatenated output as "detections".
Naming the input tensort as "input".
Building the TensorRT engine.  This would take a while...
(Use "--verbose" or "-v" to enable verbose logging.)
onnx_to_tensorrt.py:170: DeprecationWarning: Use build_serialized_network instead.
  engine = builder.build_engine(network, config)
[TensorRT] WARNING: Half2 support requested on hardware without native FP16 support, performance will be negatively affected.
[TensorRT] ERROR: 1: [calibrator.cpp::add::779] Error Code 1: Cuda Runtime (an illegal memory access was encountered)
[TensorRT] ERROR: [executionContext.cpp::commonEmitDebugTensor::1258] Error Code 1: Cuda Runtime (an illegal memory access was encountered)
[TensorRT] ERROR: [convolutionRunner.cpp::executeConv::454] Error Code 1: Cudnn (CUDNN_STATUS_EXECUTION_FAILED)
[TensorRT] INTERNAL ERROR: [defaultAllocator.cpp::free::85] Error Code 1: Cuda Runtime (an illegal memory access was encountered)
[TensorRT] INTERNAL ERROR: [defaultAllocator.cpp::free::85] Error Code 1: Cuda Runtime (an illegal memory access was encountered)
[TensorRT] INTERNAL ERROR: [defaultAllocator.cpp::free::85] Error Code 1: Cuda Runtime (an illegal memory access was encountered)
Cuda failure in file 'yolo_layer.cu' line 85: 700
Aborted (core dumped)

~~~

jkjung-avt, 2021-11-24:
Did you follow the steps in [Demo \#6: Using INT8 and DLA core](https://github.com/jkjung-avt/tensorrt_demos#demo-6-using-int8-and-dla-core) and prepare the **images for INT8 calibration**?  If yes, did you see any error related to the calibrator?

3a532028, 2021-11-25:
> Did you follow the steps in [Demo #6: Using INT8 and DLA core](https://github.com/jkjung-avt/tensorrt_demos#demo-6-using-int8-and-dla-core) and prepare the **images for INT8 calibration**? If yes, did you see any error related to the calibrator?

First of all, thank you so much for your contribution in this work, @jkjung-avt  !

And I found why the error caused.
That was because I modified the MAX_BATCH_SIZE param which both in  "yolo_to_onnx.py" & "onnx_to_tensorrt.py" as 32.
After that, I settled MAX_BATCH_SIZE = 1. So the program  ‎Quantification TensorRT INT8 successfully.

jkjung-avt, 2021-11-25:
Good to know that you've got the issue resolved.

closed 2021-12-02 by jkjung-avt