jkjung-avt/tensorrt_demos #511: How to set dynamic batch_size in Tensorrt model?

3a532028, 2021-11-24:
I want to set dynamic batch_size in tensorrt model for Triton service purpose.
How to do it?

EyGy, 2021-11-24:
Afaik tensorRT still requires fixed input batch size:
See https://github.com/jkjung-avt/tensorrt_demos/issues/25

jkjung-avt, 2021-11-24:
This discussion might be helpful: https://github.com/jkjung-avt/tensorrt_demos/issues/457

You could also check out the dynamic batch size plugin implementation by @philipp-schmidt: https://github.com/jkjung-avt/tensorrt_demos/pull/465

philipp-schmidt, 2021-11-24:
Hello everyone,

dynamic batch size requires a different implementation of the plugin. It is only the interface that changes, not the actual code / computation. I have submitted a merge request that I could not finish properly till now unfortunately. It works well with the standard yolov4 but I was seeing some errors for other variants (e.g. yolov4-tiny-3l). This suggests that some of the "dimension calculations" functions is not generalized well for all variants and calculates stuff wrongly.

ONNX unfortunately does not support all plugin implementations, so we can not just use my plugin implementation and settings from yolov4-triton-tensorrt repo. ONNX rejects that. Dynamic batching works there when I implement the network layer by layer in TensorRT without ONNX.

You can have a look at the changes I made in my PR and double check them. Jkjung didn't have the time yet which is absolutely fair.

philipp-schmidt, 2021-11-24:
@EyGy that's not true, my repo shows otherwise. I can use dynamic input size. It is the ONNX loader which expects fixed size for plugin layers or the most recent plugin interface.

I don't have the details, but the main difference is the "mode" that you load networks in TensorRT. It is a flag of the builder in the code.

philipp-schmidt, 2021-11-24:
I think there is a mode where you can have a dynamic batch (!) size but are limited to static input dimension (TensorRT will just assume the batch size for a static input definition). Unfortunately ONNX rejects that mode.

EyGy, 2021-11-24:
Very interesting, so i guess i had it wrong...
Thanks for the clarification!

closed 2021-12-02 by jkjung-avt