jkjung-avt/tensorrt_demos #512: assert size % 7 == 0 AssertionError

dexception, 2021-11-26:
```
root@asd-TUF-GAMING-FX504GM-FX80GM:/opt/github/tensorrt_demos# python3 trt_yolo.py --image dog.jpg -m yolov3-tiny-416
[TensorRT] WARNING: TensorRT was linked against cuBLAS 10.2.0 but loaded cuBLAS 10.1.0
<tensorrt.tensorrt.ICudaEngine object at 0x7f44d0dca848>
[TensorRT] WARNING: TensorRT was linked against cuBLAS 10.2.0 but loaded cuBLAS 10.1.0
Traceback (most recent call last):
  File "/opt/github/tensorrt_demos/utils/yolo_with_plugins.py", line 296, in __init__
    allocate_buffers(self.engine)
  File "/opt/github/tensorrt_demos/utils/yolo_with_plugins.py", line 221, in allocate_buffers
    assert size % 7 == 0
AssertionError

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "trt_yolo.py", line 108, in <module>
    main()
  File "trt_yolo.py", line 96, in main
    trt_yolo = TrtYOLO(args.model, args.category_num, args.letter_box)
  File "/opt/github/tensorrt_demos/utils/yolo_with_plugins.py", line 298, in __init__
    raise RuntimeError('fail to allocate CUDA resources') from e
RuntimeError: fail to allocate CUDA resources
Exception ignored in: <bound method TrtYOLO.__del__ of <utils.yolo_with_plugins.TrtYOLO object at 0x7f44d0dc2f28>>
Traceback (most recent call last):
  File "/opt/github/tensorrt_demos/utils/yolo_with_plugins.py", line 305, in __del__
    del self.outputs
AttributeError: outputs

```

jkjung-avt, 2021-11-27:
Did you see any error when you built the TensorRT ("yolov3-tiny-416.trt") engine?

Otherwise, what are the shapes and sizes of the TensorRT output tensors?

dexception, 2021-12-05:
[image]

utils/yolo_with_plugins.py
is giving the following output:

[image]

jkjung-avt, 2021-12-09:
From your screenshot,

* 43095 = 3 * 5 * 13 * 13 * 17

But the expected output tensor size should be:

* (13 * 13) * 3 * 7 + (26 * 26) * 3 * 7 = 3 * 7 * 13 *13 * 17

I'm guessing that: (a) you are using a custom model with `classes=1`; (b) the TensorRT model (yolov4-tiny-416.trt) is missing the yolo_layer plugins ("YoloLayer_TRT" layers).

Are you using the latest code in this repo?  What code modification have you done?

dexception, 2021-12-17:
Using the latest repo and also the default model with 80 classes.

jkjung-avt, 2021-12-17:
> Using the latest repo and also the default model with 80 classes.

I'm not able to reproduce this problem.  I ran the code on several difference platforms (Jetson and x86).  It all worked OK.  So I don't know what the problem is for you...

jkjung-avt, 2021-12-29:
Closed as of "not reproducible".

closed 2021-12-29 by jkjung-avt