jkjung-avt/tensorrt_demos #516: Can I install mtcnn samples in ubentu 18.04?

LUOQING1994, 2021-12-16:


LUOQING1994, 2021-12-16:
My environment：ubentu 18.04、TensorRt 6.0.x、cuda 10.1
When I build the Cython code. the following error is reported：
[image]

jkjung-avt, 2021-12-16:
@LUOQING1994  Please refer to [README_x86.md](https://github.com/jkjung-avt/tensorrt_demos/blob/master/README_x86.md).

To recap, please find whether your TensorRT 6.0.x installation is located.

For example, I put the "/usr/local/TensorRT-7.1.3.4" path into my library and include paths since that's where I've installed TensorRT on my x86 PC.  Your problem should be fixed once you replace them with the correct paths for your own environment.

https://github.com/jkjung-avt/tensorrt_demos/blob/2e31c5deb42cf510030e8f6571b5a49671d70996/setup.py#L8-L12

https://github.com/jkjung-avt/tensorrt_demos/blob/2e31c5deb42cf510030e8f6571b5a49671d70996/setup.py#L24-L32

closed 2021-12-24 by jkjung-avt