jkjung-avt/tensorrt_demos #517: libnvinfer.so.8: cannot open shared object file: No such file or directory

akashAD98, 2021-12-18:
im using tensorrt 8.2.1.8 ,cuda 11.5 on ubantu 20.04 system 

 im getting this issue
 ImportError: libnvinfer.so.8: cannot open shared object file: No such file or directory

[image]

even i have specified path in .bashrc

[image]

& inside plugins makefile

[image]

jkjung-avt, 2021-12-19:
> export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:</home/ubuntu/TensorRT-8.2.1.8/lib>

Could you try to remove the angle brackets ("<" & ">") in this line?  And make sure the path is correct: "/home/ubuntu/TensorRT" or "/home/ubuntu/TensorRT-8.2.1.8".

akashAD98, 2021-12-21:
sure ill try it

akashAD98, 2021-12-24:
@jkjung-avt after  changing path & rebooting of  my server its working fine thanks

akashAD98, 2022-09-12:
[image]

akashAD98, 2022-09-12:
again im getting the same issue, i checked all my path still getting issue.

akashAD98, 2022-09-12:
is this problem with tensoort or cuda?

jkjung-avt, 2022-09-12:
> libnvinfer.so.8: cannot open shared object file: No such file or directory

This looks like a problem of TensorRT library installation.

You need to make sure: (a) you have installed TensorRT properly, and (b) "libnvinfer.so.8" is on your [shared library search paths](https://unix.stackexchange.com/questions/22926/where-do-executables-look-for-shared-objects-at-runtime).

akashAD98, 2022-09-12:
@jkjung-avt i uninstalled tensorrt & installed it again & also changed name into simple Tensorrt & it worked.

closed 2022-09-12 by akashAD98

akashAD98, 2022-09-12:
@jkjung-avt thanks for the quick response. I'm grateful to you