jkjung-avt/tensorrt_demos #519: About an inference problem

chakloong883, 2021-12-23:
I read the code, and I pay attention to YOLO inference process. I want to find out how anchor settings would affect the inference of final bounding boxes. But I don't see this process in the code, could you please tell me where it is.

jkjung-avt, 2021-12-23:
"onnx_to_tensorrt.py" would read anchor values from your .cfg file, and save those anchor values in the yolo_layer plugins.  You could refer to the relevant source code below.

https://github.com/jkjung-avt/tensorrt_demos/blob/2e31c5deb42cf510030e8f6571b5a49671d70996/yolo/plugins.py#L24-L25

https://github.com/jkjung-avt/tensorrt_demos/blob/2e31c5deb42cf510030e8f6571b5a49671d70996/yolo/plugins.py#L135-L136

chakloong883, 2021-12-23:
Thank you so much!

chakloong883, 2021-12-23:
> "onnx_to_tensorrt.py" would read anchor values from your .cfg file, and save those anchor values in the yolo_layer plugins. You could refer to the relevant source code below. [...]
Thank you so much!

closed 2022-01-05 by jkjung-avt