jkjung-avt/tensorrt_demos #522: ERROR: failed to build the TensorRT engine!

ddinhtuann, 2022-01-04:
I try to convert onnx to tensorrt int8 but got this error bellow,would some one help me up?[image]

jkjung-avt, 2022-01-05:
I myself have never encountered this error before.

This probably does not help too much.  But anyway, this is what TensorRT header file ("NvInferRuntimeCommon.h") says about Error Code 2:

```
    //!
    //! A non-recoverable TensorRT error occurred. TensorRT is in an invalid internal state when this error is
    //! emitted and any further calls to TensorRT will result in undefined behavior.
    //!
    kINTERNAL_ERROR = 2,
```

ddinhtuann, 2022-01-05:
Thanks, I found why the error caused: DLA_core was wrong type and I fixed it. I finished built the tensort engine but when I run trt_yolo.py through vscode remote ssh, my jetson power off even I set maxn nvpmodel, jetson_clocks. I dont show the frames with cv2_imshow(), i just print the fps and write those frames to .avi file. Do you have any ideal?

jkjung-avt, 2022-01-05:
Are you using a proper AC-DC adapter to power the Jetson system?

ddinhtuann, 2022-01-05:
Yes, I am using a AC-DC adapter to supply 5V-4A via Power Jack

closed 2022-01-05 by ddinhtuann