jkjung-avt/tensorrt_demos #523: How to Convert onnx to tensorrt float32?

num-doc, 2022-01-05:
I don't know how to set it, it's not reflected in your code handling.
Please take a look sometime.
Demo #5: YOLOv4

jkjung-avt, 2022-01-05:
For building FP32 TensorRT engines for the yolo models, you could modify the source by yourself.  More specifically, remove the following 2 lines of code.

https://github.com/jkjung-avt/tensorrt_demos/blob/2e31c5deb42cf510030e8f6571b5a49671d70996/yolo/onnx_to_tensorrt.py#L138

https://github.com/jkjung-avt/tensorrt_demos/blob/2e31c5deb42cf510030e8f6571b5a49671d70996/yolo/onnx_to_tensorrt.py#L150

num-doc, 2022-01-05:
First of all, thank you so much for your contribution in this work,@jkjung-avt .
And then,
My problem has been solved. Thank you again!

closed 2022-01-05 by jkjung-avt