jkjung-avt/tensorrt_demos #524: How come we have low FPS even after optimizing the models

abhinav0927, 2022-01-06:
Hi

I was going to follow your project and then read the FPS you have mentioned after optimizing the models. How come we are facing low FPS even after optimizing the models. ?

jkjung-avt, 2022-01-07:
What platform are you testing the code with?  What FPS number are you expecting?  And what's the result?

abhinav0927, 2022-01-07:
Hi, I am planning to test it on Jetson Nano. I havent tested it but was going through your documentation and realized that you mentioned FPS to be around 10 which I feel is a bit low. I am expecting the FPS to be around 25-30. Do you think is that possible?

I am already getting 30-32FPS with SSD Mobilenet models but I feel SSD detection accuracy is a bit low and it cannot perform detection on smaller objects, thats why I want to switch to other models like Yolo

jkjung-avt, 2022-01-07:
Please check out the following posts about my opinions on FPS.  If FPS is really important to your application, you might consider adopting DeepStream instead.

* https://github.com/AlexeyAB/darknet/pull/5453#issuecomment-663593512
* https://github.com/AlexeyAB/darknet/pull/5453#issuecomment-665105334

closed 2022-01-17 by jkjung-avt