jkjung-avt/tensorrt_demos #526: Error when autostart on boot with crontab

Davidgo123, 2022-01-12:
Hello,

has anyone a Idea, why i get this Error?
When i start the Script Manuell in the shell, it works...

**Command in Crontab:**
@reboot source $HOME/.bashrc && /home/david/Bachelorarbeit/tensorrt/autostart_BA.sh >/home/david/Bachelorarbeit/tensorrt/BA.log 2>&1

**Script for startup:**
python /home/david/Bachelorarbeit/tensorrt/trt_yolo.py -m yolov4-tiny-ampelpilot-640 --onboard 0 --width 1000 --height 800

**Error Message:**
GST_ARGUS: Running with following settings:
   Camera index = 0
   Camera mode  = 2
   Output Stream W = 1920 H = 1080
   seconds to Run    = 0
   Frame Rate = 29.999999
GST_ARGUS: Setup Complete, Starting captures for 0 seconds
GST_ARGUS: Starting repeat capture requests.
CONSUMER: Producer has connected; continuing.
[TensorRT] INFO: [MemUsageChange] Init CUDA: CPU +202, GPU +0, now: CPU 245, GPU 2481 (MiB)
[TensorRT] INFO: Loaded engine size: 28 MB
[TensorRT] INFO: [MemUsageSnapshot] deserializeCudaEngine begin: CPU 273 MiB, GPU 2534 MiB
[TensorRT] WARNING: Using an engine plan file across different models of devices is not recommended and is likely to affect performance or even cause errors.
[TensorRT] INFO: [MemUsageChange] Init cuBLAS/cuBLASLt: CPU +158, GPU +272, now: CPU 435, GPU 2837 (MiB)
[TensorRT] INFO: [MemUsageChange] Init cuDNN: CPU +239, GPU +413, now: CPU 674, GPU 3250 (MiB)
[TensorRT] INFO: [MemUsageChange] Init cuBLAS/cuBLASLt: CPU +0, GPU +0, now: CPU 676, GPU 3250 (MiB)
[TensorRT] INFO: [MemUsageSnapshot] deserializeCudaEngine end: CPU 676 MiB, GPU 3250 MiB
[TensorRT] INFO: [MemUsageSnapshot] ExecutionContext creation begin: CPU 648 MiB, GPU 3223 MiB
[TensorRT] INFO: [MemUsageChange] Init cuBLAS/cuBLASLt: CPU +0, GPU +0, now: CPU 648, GPU 3223 (MiB)
[TensorRT] INFO: [MemUsageChange] Init cuDNN: CPU +0, GPU +0, now: CPU 648, GPU 3223 (MiB)
[TensorRT] INFO: [MemUsageSnapshot] ExecutionContext creation end: CPU 648 MiB, GPU 3266 MiB
: cannot connect to X server
(Argus) Error InvalidState: Argus client is exiting with 2 outstanding client threads (in src/rpc/socket/client/ClientSocketManager.cpp, function recvThreadCore(), line 357)
CONSUMER: Done Success
WARNING Argus: 6 client objects still exist during shutdown:
        547820181264 (0x7f40000ee0)
        547825157056 (0x7f40000c70)
        547825157408 (0x7f40001ff0)
        547825163520 (0x7f40002100)
        547825164816 (0x7f40003ce0)
        547825172904 (0x7f40003df8)
-------------------------------------------------------------------
PyCUDA ERROR: The context stack was not empty upon module cleanup.
-------------------------------------------------------------------
A context was still active when the context stack was being
cleaned up. At this point in our execution, CUDA may already
have been deinitialized, so there is no way we can finish
cleanly. The program will be aborted now.
Use Context.pop() to avoid this problem.
-------------------------------------------------------------------
Aborted (core dumped)

jkjung-avt, 2022-01-13:
> : cannot connect to X server

This message shows:

1. "cv2" tries to connect to X server to open a new window (for displaying inference results),
2. the operation fails, probably because X server has not started yet.

The problem could probably be solved by adding some delay before starting the script.

Davidgo123, 2022-01-13:
thanks for edit and help!

I testet with delay of 1 min in Crontab and direkt in the python script, but got the same error message ...
Any other ideas...?

jkjung-avt, 2022-01-14:
Please refer to https://stackoverflow.com/questions/66290650/unable-to-run-a-python-program-with-cron-cannot-connect-to-x-server.

Davidgo123, 2022-01-16:
**solved the Problem:**

Used `crontab -e` insteed of `sudo crontab -e`.

And specify Display in Crontab:
`@reboot sleep 10 && export DISPLAY=:0 && /home/david/Bachelorarbeit/tensorrt/autostart_BA.sh >/home/david/Bachelorarbeit/tensorrt/BA.log 2>&1`

can be closed.

closed 2022-01-16 by Davidgo123