jkjung-avt/tensorrt_demos #529: Could not install ONNX=1.4.1 on jetson xavier

ASingh245, 2022-01-25:
when I execute the below command I am getting below error,
sudo pip3 install onnx==1.4.1
WARNING: The directory '/home/xavier2/.cache/pip' or its parent directory is not owned or is not writable by the current user. The cache has been disabled. Check the permissions and owner of that directory. If executing pip with sudo, you should use sudo's -H flag.
Collecting onnx==1.4.1
  Downloading onnx-1.4.1.tar.gz (2.9 MB)
     |████████████████████████████████| 2.9 MB 10.2 MB/s
  Preparing metadata (setup.py) ... done
Requirement already satisfied: protobuf in /usr/local/lib/python3.7/dist-packages (from onnx==1.4.1) (3.19.3)
Requirement already satisfied: numpy in /usr/local/lib/python3.7/dist-packages (from onnx==1.4.1) (1.19.4)
Requirement already satisfied: six in /usr/local/lib/python3.7/dist-packages (from onnx==1.4.1) (1.14.0)
Requirement already satisfied: typing>=3.6.4 in /usr/local/lib/python3.7/dist-packages (from onnx==1.4.1) (3.7.4.3)
Requirement already satisfied: typing-extensions>=3.6.2.1 in ./.local/lib/python3.7/site-packages (from onnx==1.4.1) (3.10.0.2)
Building wheels for collected packages: onnx
  Building wheel for onnx (setup.py) ... error
  ERROR: Command errored out with exit status 1:
   command: /usr/bin/python3 -u -c 'import io, os, sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-gmp8dzqm/onnx_9ad126445f504d9abe902706c71cb1b1/setup.py'"'"'; __file__='"'"'/tmp/pip-install-gmp8dzqm/onnx_9ad126445f504d9abe902706c71cb1b1/setup.py'"'"';f = getattr(tokenize, '"'"'open'"'"', open)(__file__) if os.path.exists(__file__) else io.StringIO('"'"'from setuptools import setup; setup()'"'"');code = f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' bdist_wheel -d /tmp/pip-wheel-yph89h6c
       cwd: /tmp/pip-install-gmp8dzqm/onnx_9ad126445f504d9abe902706c71cb1b1/
  Complete output (131 lines):
  fatal: not a git repository (or any of the parent directories): .git
  running bdist_wheel
  running build
  running build_py
  running create_version
  running cmake_build
  -- Build type not set - defaulting to Release
  -- The C compiler identification is GNU 7.5.0
  -- The CXX compiler identification is GNU 7.5.0
  -- Detecting C compiler ABI info
  -- Detecting C compiler ABI info - done
  -- Check for working C compiler: /usr/bin/cc - skipped
  -- Detecting C compile features
  -- Detecting C compile features - done
  -- Detecting CXX compiler ABI info
  -- Detecting CXX compiler ABI info - done
  -- Check for working CXX compiler: /usr/bin/c++ - skipped
  -- Detecting CXX compile features
  -- Detecting CXX compile features - done
  -- Found Protobuf: /usr/local/lib/libprotobuf.so (found version "3.8.0")
  CMake Warning at CMakeLists.txt:356 (find_package):
    By not providing "Findpybind11.cmake" in CMAKE_MODULE_PATH this project has
    asked CMake to find a package configuration file provided by "pybind11",
    but CMake did not find one.

    Could not find a package configuration file provided by "pybind11"
    (requested version 2.2) with any of the following names:

      pybind11Config.cmake
      pybind11-config.cmake

    Add the installation prefix of "pybind11" to CMAKE_PREFIX_PATH or set
    "pybind11_DIR" to a directory containing one of the above files.  If
    "pybind11" provides a separate development package or SDK, be sure it has
    been installed.

  --
  -- ******** Summary ********
  --   CMake version         : 3.19.4
  --   CMake command         : /usr/local/bin/cmake
  --   System                : Linux
  --   C++ compiler          : /usr/bin/c++
  --   C++ compiler version  : 7.5.0
  --   CXX flags             :  -Wnon-virtual-dtor
  --   Build type            : Release
  --   Compile definitions   :
  --   CMAKE_PREFIX_PATH     :
  --   CMAKE_INSTALL_PREFIX  : /usr/local
  --   CMAKE_MODULE_PATH     :
  --
  --   ONNX version          : 1.4.1
  --   ONNX NAMESPACE        : onnx
[... 184 lines cut ...]
        self.run_command('cmake_build')
      File "/usr/lib/python3.7/distutils/cmd.py", line 313, in run_command
        self.distribution.run_command(command)
      File "/usr/lib/python3.7/distutils/dist.py", line 985, in run_command
        cmd_obj.run()
      File "/tmp/pip-install-gmp8dzqm/onnx_9ad126445f504d9abe902706c71cb1b1/setup.py", line 197, in run
        subprocess.check_call(build_args)
      File "/usr/lib/python3.7/subprocess.py", line 363, in check_call
        raise CalledProcessError(retcode, cmd)
    subprocess.CalledProcessError: Command '['/usr/local/bin/cmake', '--build', '.', '--', '-j', '8']' returned non-zero exit status 2.
    ----------------------------------------
  Rolling back uninstall of onnx
  Moving to /home/xavier2/.local/bin/backend-test-tools
   from /tmp/pip-uninstall-e0rovw33/backend-test-tools
  Moving to /home/xavier2/.local/bin/check-model
   from /tmp/pip-uninstall-e0rovw33/check-model
  Moving to /home/xavier2/.local/bin/check-node
   from /tmp/pip-uninstall-e0rovw33/check-node
  Moving to /home/xavier2/.local/lib/python3.7/site-packages/onnx-1.10.2.dist-info/
   from /home/xavier2/.local/lib/python3.7/site-packages/~nnx-1.10.2.dist-info
  Moving to /home/xavier2/.local/lib/python3.7/site-packages/onnx/
   from /home/xavier2/.local/lib/python3.7/site-packages/~nnx
ERROR: Command errored out with exit status 1: /usr/bin/python3 -u -c 'import io, os, sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-gmp8dzqm/onnx_9ad126445f504d9abe902706c71cb1b1/setup.py'"'"'; __file__='"'"'/tmp/pip-install-gmp8dzqm/onnx_9ad126445f504d9abe902706c71cb1b1/setup.py'"'"';f = getattr(tokenize, '"'"'open'"'"', open)(__file__) if os.path.exists(__file__) else io.StringIO('"'"'from setuptools import setup; setup()'"'"');code = f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /tmp/pip-record-k_za2yiv/install-record.txt --single-version-externally-managed --compile --install-headers /usr/local/include/python3.7/onnx Check the logs for full command output.

Please Advice

jkjung-avt, 2022-01-30:
Have you installed "protobuf"?

jkjung-avt, 2022-02-21:
Close due to no response.

closed 2022-02-21 by jkjung-avt

jkjung-avt, 2022-05-06:
https://github.com/jkjung-avt/tensorrt_demos/issues/539#issuecomment-1119218462