jkjung-avt/tensorrt_demos #533: How do i stop the tensorrt engine?

chakloong883, 2022-03-02:
I want to develop on qt and design a pause function, I want to know how to stop tensorrt's inference engine to avoid errors when continuing detection. Looking forward to your answer

jkjung-avt, 2022-03-02:
Every time you call "execute()" on a TensorRT engine, it would process 1 batch of sample (e.g. 1 image).

In other words, if you don't call "execute()" continuously, the engine will not continue to do inference or detection.

Take [trt_yolo.py](https://github.com/jkjung-avt/tensorrt_demos/blob/67c4eb0f5e8a46656eb65073e5685dc7654498d8/trt_yolo.py) for example, I have wrapped the "execute()" call inside TrtYOLO's "detect()" function.  I call this "detect()" in a while loop so it does detection continuously.

https://github.com/jkjung-avt/tensorrt_demos/blob/67c4eb0f5e8a46656eb65073e5685dc7654498d8/trt_yolo.py#L69

If you understand the source code, you could just put this "detect()" call at the proper place.  There's no need for a pause function.

chakloong883, 2022-03-04:
> Every time you call "execute()" on a TensorRT engine, it would process 1 batch of sample (e.g. 1 image).
>
> In other words, if you don't call "execute()" continuously, the engine will not continue to [...]

Thank you so much! I learned a lot, you can close the issue.

closed 2022-03-04 by jkjung-avt